PiSupply/PiJuice #477: RTC cannot be accessed via hwclock

teakholz, 2020-03-10:
Hey Guys, (:

first of all: I'm pretty new to this field. I'm writing my master thesis in which I shall built a small station that will be installed in the field. There it shall capture pictures for a short period and send them via mobile internet.

For that purpose I'm using the sixfab LTE base shield and the pijuice zero (with 12000 mAh battery). And as far as I can tell, these two shields do not use the same pins.

All features that I need work fine (mobile Internet and wakeup) except for the fact, that I cannot access the time of the PiJuice, which is a problem for my project.

I searched now for quiet some time in the Internet for a solution. However, the biggest problem right now is, that many people had the same issue, but the solution that worked for them, does not work for me. 
I tried to add the lines "dtoverlay=i2c-rtc,ds1339" and "dtoverlay=i2c-rtc,ds1339" to the config.txt file. I switched off "Wakeup enabled" setting in the pijuice_cli and I checked that I'm using 50 for the EEPROM address.

Two times the 68 changed to UU after boot when I called "i2cdetect -y 1", but after a new reboot, it displayed again 68, which just confused me even more.
At the moment I don't know what else I could do and I lack a deep understanding of the architecture and communication of the parts.

Any ideas/ suggestions would be very welcome.

Best regards 

teakholz

tvoverbeek, 2020-03-10:
Each HAT has an eeprom at address 0x50 on GPIO pins  27 and 28 (GPIO 0 and 1). With two HATs the eeprom reading will not work (depends on who wins).
For the PiJuice, change the eeprom address to 0x52 so it does not interfere with the LTE hat.
Use the `dtoverlay=i2c-rtc,ds1307` in /boot/config.txt.
Since you have an internet connection I do not recommend to use `dtoverlay=i2c-rtc,ds1339`.
The os control of the RTC alarm will interfere with your programmed wakeup settings.
ds1307 has the same registers as the ds1339 to read/write the time, but no alarm function.
Hence you are in full control of the PiJuice RTC alarm.
With the `dtoverlay=i2c-rtc,ds1307` you should get the 'UU' in `i2cdetect -y 1` for 68.
Also `sudo hwclock -r` should work.

teakholz, 2020-03-11:
Thanks a lot tvoverbeek. It didnt work right away, but after I added "i2c-bcm2708" to the /etc/modules it worked. You really helped me a lot, as I was pretty lost yesterday.
However, there are a few questions left. To what refers the address 0x50? I'm confused as there are just 40 physical pins and when I tried to find information on the Internet yesterday, I did not know what to search for. And do the HATs use the pins 27/ 28 or 0/1 (current and ground)?
Another thing is, that the RTC of the PiJuice has 1 hour of delay. It is not such a big problem for me, but I don't understand why it has this 1 hour of delay. Is there an explanation for it?

tvoverbeek, 2020-03-11:
Since you had to add 'i2c-bcm2708' to /etc/modules you are probably using a Pi4 and not the most recent firmware.

0x50 is the I2C address of the HAT eeprom on the I2C bus. Pins 27,28 (GPIOs 0 and 1 in Broadcom numbering) are connected to the Pi I2c bus 0.
Each official HAT has an eeprom at address 0x50 connected to pins 27 and 28 (I2C bus 0) which is read at boot.
See https://github.com/raspberrypi/hats/blob/master/README.md for all the details.

External I2c bus peripherals are using I2C bus 1 (pins 3 and 5, GPIO 2 and 3)

Regarding the RTC. The RTC is kept in UTC. Raspbian/Linux/Unix always keep the time in UTC internally.
Externally they convert to local time (based on timezone and summer/standard time).
I guess you are in continental Europe (...holz) which is UTC+1 till March 29.

Suggest to update to the latest software (1.6 packages) using the usual `sudo apt-get update; sudo apt-get upgrade`). After the update install the 1.4 firmware via the GUI or CLI.
With the 1.4 firmware you can remove the "i2c-bcm2708" from /etc/modules.

teakholz, 2020-03-11:
Well, thanks again for the fast and detailed explanation. And yes, I'm from Germany. 
I will follow that link and then hopefully understand better the way the parts work and communicate.

Regarding the 'i2c-bcm2708'-problem: I'm using a Raspberry Pi 2B and I have updated the PiJuice firmware to 1.4 before first use. Is it a problem to have the 'i2c-bcm2708' added to /etc/modules? Right now I'm just happy that everything works as I want it to work but of course I wanna have a stable system.

tvoverbeek, 2020-03-11:
@teakholz The i2c-bcm2708 was the i2c kernel module for previous versions of older Raspbian.
Current Raspbian (Buster) uses the i2c-bcm2835 module. However firmware 1.3 had problems with i2c on the Pi4. Falling back on the previous i2c-bcm2708 made it work
The current firmware (1.4) works fine with the i2x-bcm2835 module.
What version of Raspbian / kernel are you using? The Pi2B is now 5 years old. Are you running an older version of Raspbian or an other OS on your 2B?

teakholz, 2020-03-12:
I'm using Raspian Buster (10) and the output of uname -a is as follows:
Linux raspberrypi 4.19.97-v7+ #1294 SMP ... armv71 GNU/Linux

I tried to add i2c-bcm2835 instead of i2c-bcm2708 to the /etc/modules but then UU changed back to 68.

teakholz, 2020-03-12:
And another question about strange behavior that just occurred: Should I enable  the "EEPROM write unprotect"? I have read in some other posts that this is necessary, when changing the EEPROM address to 52. However, when I enable this setting, it is disabled after the reboot. 
Is that supposed to be like that?

tvoverbeek, 2020-03-12:
I am still puzzled why you had to add i2c-bcm2708 to /etc/modules.
The pijuice-base installation enables i2c via raspi-config.
It consists of adding i2c-dev to /etc/modules and the line `dtoverlay=i2c-arm=on` to /boot/config.txt.
The dtoverlay line is enough to load the i2c-bcm2835  module at boot
What is your output from the following two commands:
grep i2c /boot/config.txt
lsmod | grep i2c
Mine is:
```
pi@raspberrypi:~ $ grep i2c /boot/config.txt
dtparam=i2c_arm=on
pi@raspberrypi:~ $ lsmod | grep i2c
i2c_bcm2835            16384  2
i2c_dev                20480  4
pi@raspberrypi:~ $
```
Regarding 'EEPROM write unprotect',  this setting has nothing to do with changing the eeprom address.
When write unprotect is enabled you could write to the eeprom with special software.
See e.g. https://github.com/raspberrypi/hats/tree/master/eepromutils
I know I wrote differently some time ago, but that was wrong (as you found out).
For me the 'ID EEPROM Write unprotect' setting persists over reboots.

teakholz, 2020-03-13:
Good morning,

thanks again for your help and your patience. I really appreciate it!
When I came to the office this morning and booted the RasPi  just worked fine (appearing of UU instead of 68) after it did not work all the day yesterday. And I cannot understand this chaotic behavior.

The output of `grep i2c /boot/config.txt` : 
`dtaparam=i2c_arm=on`
`#dtaparam=i2c=on`
`#dtoverlay=i2c-rtc,ds1339`
`dtoverlay=i2c-rtc,ds1307`

The commented lines are from other posts I read and that I tried in the beginning.

And the output of `lsmod | grep i2c` : 
`i2c_bcm2835         16384     1`
`i2c_bcm2708         16384     0`
`i2c_dev                  20480     2`

I guess there should be just one bcm-module? In the `/etc/modules`, I'm loading just `i2c-dev` and `i2c-bcm2708` for the moment. `i2c-bcm2835` is commented.
Regarding your confusion about me having added the `i2c-bcm2708` in order to make it work: Yesterday I started to doubt weather the addition of `i2c-bcm2708` to the `/etc/modules` really was the step that made everything work. Maybe it was just a coincidence that it worked right after changing this setting and since it worked I did not question it.  Because without changing any settings, the behavior can change after a reboot.

I'm also still confused why the "ID EEPROM Write unprotect" setting does not persist over reboot.

teakholz, 2020-03-13:
After reading about the "lsmod" command, I removed the `i2c_bcm2708` from the `/etc/modules` and, for the moment, it still works. So I guess it was indeed just a bad coincidence. 
Does it make a difference whether I reboot or poweroff and then boot the RasPi after I made some changes?

tvoverbeek, 2020-03-13:
> Does it make a difference whether I reboot or poweroff and then boot the RasPi after I made some changes?

No, changes in /boot/config.txt and /etc/modules are picked up at (re)boot.

Getting only the Sixfab LTE Hat is what you want. There can only be a single HAT recognized at boot by reading its eeprom.
The PiJuice RTC is recognized by the `dtoverlay=i2c-rtc,ds1307` line in /boot/config.txt.
The i2c-bcm2835 module is loaded because of the `dtparam=i2c-arm=on` in /boot/config.txt.
You  have a typo there `dtaparm` instead of `dtparam`. With the correct line you do not need i2c-bcm2835 in /etc/modules. This typo is probably the cause of your problems.

teakholz, 2020-03-13:
That would have been a good explanation, but unfortunately the typo happened here in the comment, when I wrote down the output from another screen. I edited the post and now it shows the real output that I get from the two commands. 

Could there have been some conflicts between `i2c_bcm2708` and `i2c_bcm2835`, since I added the first manually in `/etc/modules`?

tvoverbeek, 2020-03-13:
Yes, two drivers for the same interface. Suppose it is a chance which one wins.
Schönes Wochenende.

teakholz, 2020-03-13:
Well, it works for the moment, but I haven't verified it since I work today on a different project. Next week I will do some tests to see hopefully whether it's stable or not.
Danke sehr und ebenso.

shawaj, 2020-10-24:
@teakholz is this all working for you now? Can we close this issue?

teakholz, 2020-11-02:
Hello,
I'm sorry that I havent written again. Everything was working fine at the end. Therefore, thanks a lot for your help!
Best regards.

tvoverbeek, 2020-11-02:
Glad it works for you. Closing the issue.

closed 2020-11-02 by tvoverbeek