PiSupply/PiJuice #479: Shuts off charging when battery gets below 12volts

JVTEAM, 2020-03-22:
I have the pijuice hooked up to a remote camera that has a solar panel and a 12v battery setup.  The Solar Controller will completely turn off charging if it stops charging something for a certain amount of time.  I'm noticing that the pijuice is refusing a charge from a battery that is running below 12v.

I'm fine with undervolting.  Is there a way to force it to accept a charge until that battery is completely dead?

shawaj, 2020-03-22:
The PiJuice doesn't accept a 12v input.

Are you reducing the voltage to 5v?

JVTEAM, 2020-03-22:
yes.........
the solar controller has a usb output.  
Software, not hardware answers please.

tvoverbeek, 2020-03-22:
@JVTEAM To give a software answer I need to know your hardware setup.
Solar Array connected to PiJuice Micro-USB?
How does the 12V battery (car battery?) come into play?
Connected to what?
RPi power only via PiJuice or ???

JVTEAM, 2020-03-22:
very well.
This is the solar controller https://www.amazon.com/gp/product/B01LZZMDSQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
The solar panel is this.
https://www.amazon.com/gp/product/B00OZC19AY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
The Battery is this
https://www.amazon.com/gp/product/B0089N311K/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

The USB plugs into the solar controller USB output.  The USB plugs into the pijuice.  The pijuice is connected to the RPI.  The hip bones connected to the leg bones.

tvoverbeek, 2020-03-22:
It seems (if your solar panel link is correct) that it is only a 5W panel. After conversion to 5V it would mean max current of 1A. There will be conversion losses, so max current at USB output will be less than 1A (based on solar inut). So even with the solar panel active you will drain the battery.
In one of the answers on the Amazon page for the solar controller it says that the output is switched off when the battery voltage goes below 12V. Assume it will also switch off the the USB output.
I guess this is what you are seeing. With no power input to the PiJuice it obviously will not charge.
Can you get load information from the display on the solar controller, to see if this is what is happening?

JVTEAM, 2020-03-22:
Yes, that is what i'm seeing.  interesting that it is the controller, not the pi, that is doing this.

essentially the upper right hand display goes from a number to off once the batter is below 12V.

tvoverbeek, 2020-03-22:
I guess your best bet is to use a larger capacity solar array to be sure that the solar array has enough power to be able to run the load (RPi + Camera) and keep the 12V battery charged simultaneously.

JVTEAM, 2020-03-22:
thanks everyone, ordered a beefier panel.

tvoverbeek, 2020-03-22:
Closing the issue. Reopen it if needed.

closed 2020-03-22 by tvoverbeek