PiSupply/PiJuice #480: pijuice not working on pi4b

Jens5395, 2020-03-23:
I am trying to install an older pijuice on a pi4b. I pulled the plastic battery tab, installed the unit on the 4b with a header extender to allow for heat sinks and installed pijuice software. After rebooting I do not get the pijuice icon in the top icon bar and if I go to preferences, pijuice settings, the pi4 is doing something (hour glass for a few seconds) but produces no pijuice menu.
I tried the dtoverlay=i2c-bcm2708 in /boot/config.txt as mentioned here in other posts but ther was no change in behaviour after rebooting.
Probably something stupidly simple but I have no clue what it could be.

shawaj, 2020-03-23:
Are you using Raspbian OS?

Do you have anything else attached to the Pi?

Jens5395, 2020-03-23:
Yes, the official Raspbian os as far as I know (got the raspberry logo and stuff)
There is a 2 TB solid state hard drive attached via USB3 and the unit is attached to a network via ethernet

shawaj, 2020-03-23:
And how did you install the software? Did you install both the GUI and base packages?

Jens5395, 2020-03-23:
I did an apt-get install pijuice-gui
I did NOT install the base package as I thought that it was install one OR the other.
I will give that a try right now!

Jens5395, 2020-03-23:
tried to install pijuice-base but get a 'already newest version' message (ver 1.6) 
I also get a message saying that "pijuice-base set to manually installed"

shawaj, 2020-03-23:
If you just install the GUI package it should install the base package for your automatically.

Do you get the PiJuice icon in the start menu?

Jens5395, 2020-03-23:
Yes, screengrab attached. Note there is no battery icon in the top right .

[image]

Jens5395, 2020-03-23:
Had a look-see at the properties of 'pijuice settings' and got this (zero file size)
[image]

shawaj, 2020-03-23:
The settings screenshot you sent was for Raspberry Pi Configuration.

Do you get the same issue if you are using a display directly attached rather than using VNC?

Jens5395, 2020-03-23:
My bad, slipped a line down in the menu. If I click on the pijuice settings I get the hour glass that disapears after a bit with no other action (ie no menus)

Yes, behaviour is identical. Note that I did not do a reboot though - just disconnected the pi4 from power/network and drive and hooked it up to keyboard/mouse and monitor. Pijuice provided the power

Jens5395, 2020-03-23:
Question: 
I am gradually working through previous issues and found a reference to running
/usr/bin/pijuice_tray.py
this completes with a 'permission denied' error
I dug a bit further and it turns out I have no group 'pi'. Does pijuice require a user pi or a group pi or anything else special ?

Jens5395, 2020-03-23:
Seems that this was a dead end. There was in fact a group pi but I was not a member. There is no user pi. I added myself to the group pi but nothing changed :(
/usr/bin/pijuice_tray.py still ends with 'permission denied'

Jens5395, 2020-03-23:
I am still not clear if this might not be a permissions issue.
If I run pijuice_cli i get "No pijuice found or failed to use I2C bus. check if I2C is enabled"
When I check, I2C is definitively enabled.

tvoverbeek, 2020-03-23:
@Jens5395 First things first. Let  us check if I2C communication is working.
What is the output of `i2cdetect -y 1` when running with your userid?
If you get an error try with `sudo i2cdetect -y 1'.
Also what is the output from `cat /boot/config.txt | grep i2c`?
Run these commands in a terminal window/console/ssh session.
Here are the outputs on a working system:
```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ cat /boot/config.txt|grep i2c
dtparam=i2c_arm=on
pi@raspberrypi:~ $ 
```

Jens5395, 2020-03-23:
sudo i2cdetect -y 1 results in an output identical to the example
cat /boot/config.txt|grep i2c produces:
dtparam=i2c_arm=0n
#dtoverlay=i2c = bcm2708

The second line is the leftover from my previous attempts to get things going

tvoverbeek, 2020-03-23:
OK, I2C is working.

Problem is your username/userid not belonging to the groups i2c and pijuice
The installation script does this for userid pi (or userid 1000), but not for other users.
I assume on your system a new user was added.

Try this:
`sudo addgroup <yourusername> i2c`
After this you should be able to run i2cdetect without sudo.
If not, then we have a different problem.
Next `sudo addgroup <yourusername> pijuice`
Then try to run /usr/bin/pijuice_tray.py in a terminal window, so you can see possible error messages.
Let me know how it goes.

Jens5395, 2020-03-23:
Running i2cdetect without sudo still got a 'permission denied' error as does running /usr/bin/pijuice_tray.py

Jens5395, 2020-03-23:
A quick update: I did a reboot and I am now seeing the battery icon. clicking on it produces no action.
I also can run 'preferences/pijuice settings. I get the settings menu with a 'data corrupted' error message. Various functions 'seem' to work but I am not sure what I am supposed to see ... hence 'seem' in quotes.
Things seem to be heading the right direction ....
Running sudo /usr/bin/pijuice_tray.py generates some deprecation warnings and a stream of 'data corrupted' and refresh errors.

Jens5395, 2020-03-23:
correction re i2cdetect and /usr/bin/pijuice_tray.py - after reboot, both run under the normal user
/usr/bin/pijuice_tray.py generates refresh errors and data corrupted errors.
Sorry that I did not try a reboot earlier!

tvoverbeek, 2020-03-23:
Probably you are running an older version of the software.
What version does `dpkg -l pijuice-base` show?
Versions before 1.6 had problems on the Pi4.
Reinstate the `dtoverlay=i2c-bcm2708` in /boot/config.txt, reboot, and try again.

Jens5395, 2020-03-23:
Wooooo Hoooooo .... we have lift-off
Reinstating the dtoverlay=i2c-bcm2708 line fixed the corrupt data error!
pijuice-base is version 1.6

Thank you VERY much for your help and your incredible patience with me!!!!

tvoverbeek, 2020-03-23:
One more thing.
In the GUI check which version of the firmware are you running.
If it is 1.3, then you should upgrade to 1.4 (GUI should say new firmware is available).
After upgrading to the 1.4 firmware you can remove/comment out the `dtoverlay=i2c-bcm2708` in /boot/config.txt again.

Jens5395, 2020-03-23:
I am not sure where I would see which firmware is running. I did manage the folowing by right clicking on the battery icon:

[image]

There is nothing about new firmware.

Jens5395, 2020-03-23:
Found it, updating now ....

Jens5395, 2020-03-23:
upgraded fine, commented out line, rebooted .... perfect !!!!

tvoverbeek, 2020-03-23:
OK, to close the issue?

Jens5395, 2020-03-23:
please! ... and thanks again!

closed 2020-03-23 by Jens5395

tvoverbeek, 2020-03-23:
Glad we got it working. Will try to document this case:  other user than the default pi

shawaj, 2020-03-23:
Not sure if this matters but noticed in am above comment a typo

> sudo i2cdetect -y 1 results in an output identical to the example
> cat /boot/config.txt|grep i2c produces:
> dtparam=i2c_arm=0n
> #dtoverlay=i2c = bcm2708
> 
> The second line is the leftover from my previous attempts to get things going

You have used a zero instead of a capital O in this line:

`dtparam=i2c_arm=0n`

Jens5395, 2020-03-23:
No, it doesn't make a difference as this is an output from the pi and not an input that I typed.
I am amazed that anyone would catch anything like that.
The problem is fixed so nothing else matters :)
Thanks!

shawaj, 2020-03-23:
No worries