PiSupply/PiJuice #481: How long will the BP7X last?

mikcatta, 2020-03-23:
If the PiJuice does an immediate shutdown and it just sits there?  Will the PiJuice start the Pi up from cold after months (or years) of inactivity?

Our engineer has tried to simulate this by removing the battery, but we have found that on cold start it needs SW1 to be pressed to get things going, which isn't very helpful when the devices are remote?

tvoverbeek, 2020-03-23:
Have you enabled 'No battery turn on'?
This setting is non-volatile.
WIthout a battery, applying power to the PiJuice will  then start the Pi.
Do not know if a completely discharged battery is detected as 'No battery'

mikcatta, 2020-03-24:
Firmware 1.4.
With 'No battery turn on' enabled, it will start automatically without the battery. 
However, if I then replace the battery, plug the 5v into the PiJuice, it doesn't start. (Battery is charged to 100%). That seems a little crazy to me. Is that the same with your tests?
We have potential for 100's of those, but without the ability to power back up from a discharged state (or prove that it would), we can't go ahead. Is there any way to prove otherwise?

tvoverbeek, 2020-03-24:
Yes, when inserting a charged battery the PiJuice MCU starts and then applying power to the Pijuice will not wake up the RPi.
If you need that capability look for another product (my personal opinion)

mikcatta, 2020-03-24:
Can I just be sure that you are saying if the PiJuice battery runs down (by sitting idle), it will not start the Pi automatically when power is restored?

tvoverbeek, 2020-03-24:
The requirement is that the PiJuice MCU remains running. When the PiJuice switches off the 5V to the Pi it goes to a low power mode. The battery can certainly keep the MCU alive for several days/weeks (depending on capacity).
But several months/years will not work. When the MCU shuts down, no more wake up when power is applied.

ryanteck, 2020-03-26:
Hi @mikcatta 

What would be triggering the start, is it in the sense of if after a few months that they start up from cold without the button being pressed when they start chargng again from power source?

mikcatta, 2020-03-26:
If it was "off" for a period of months, the trigger would be that power is re-applied. As these are in portable buildings located across the UK, there would not necessarily be a person to press the button,

@tvoverbeek said that "when the MCU shuts down, it would not wake up when power is applied", which is really concerning me.

ryanteck, 2020-03-26:
There's two ways around this essentially.

The first off the top of my head is that instead applying power via the Micro USB Port on the Raspberry Pi instead will turn the Pi Juice on at the same time. However if it's where the Pi Juice is shutting the units down while they have power applied this may not be the best option.

However if I remember right there may be a setting for the Pi Juice to power up the Pi when it detects power. I'll test this out shortly to see.

mikcatta, 2020-03-26:
We definitely need that "UPS" functionality, which I understand we would loose by plugging directly into the Pi.

Under normal circumstances, re-apply power to the PiJuice does cause the Pi to boot up. However, if it's left for several months (to the point where the Pijuice battery is flat), I am being told that this won't happen, which is my biggest concern.

shawaj, 2020-03-26:
@mikcatta that should work. It would be the same as for an empty battery.

mikcatta, 2020-03-26:
So are you saying that the advice I've been given above, is wrong? 

If we remove the PiJuice battery and re-insert it a few seconds later (with the Pi shutdown obviously), applying power to the PiJuice doesn't cause the Pi to start up. Under normal circumstances it does.

shawaj, 2020-03-26:
It would be different if plugged into the Pi USB or PiJuice USB. I'm a bit confused exactly what you are proposing to do.

mikcatta, 2020-03-26:
Nope, we plug 5v into the PiJuice. 

My question is "what would happen if the PiJuice & Pi where off for a period of several months". Would re-applying power to the PiJuice a) Start up the Pi normally or b) Charge the PiJuice, but not start the Pi until SW1 was pressed?

ryanteck, 2020-03-26:
@mikcatta Is there a reason you couldn't power via the Pi's USB Port? You mention it'd loose UPS functionality?

mikcatta, 2020-03-26:
I have tried it here, and if I power via the Pi's USB port, it doesn't perform a shutdown on power-loss. That is pretty critical. Are you saying that it should?

shawaj, 2020-03-26:
@mikcatta it just immediately turns off? Or it doesn't turn off at all?

tvoverbeek, 2020-03-26:
@mikcatta You can do the following:
Power via the RPi USB.
This will run  the RPi and charge the PiJuice battery.
In the Settings connect the 'No power' event to SYS_FUNC_HALT_POW_OFF
When the 5V disappears (detection takes a few seconds to be sure it is not a quick power dip) the Pi will shut down and after 30 seconds the PiJuice switches off its 5V supply to the Pi and go to low power mode (Red Pi LED goes off).
If the 5V reappears after a few months the Pi and PiJuice will start again.
Note there is a problem when the 5V reappears during the 30 seconds wait, then the Pi will not restart.

ryanteck, 2020-03-26:
I'm also fairly certain this should work too as I've done similar myself with a project.

mikcatta, 2020-03-26:
That certainly sounds better, I am testing it now to make sure I understand.

"_Note there is a problem when the 5V reappears during the 30 seconds wait, then the Pi will not restart._" Is that something that will be fixed? I wouldn't say it happens often, but I'd rather it didn't happen at all.

shawaj, 2020-03-27:
It's in the pipeline to be fixed but we don't currently have a timeline

mikcatta, 2020-03-27:
We are slowly ramping up (about 10 pieces presently). It would be good to have this resolved and solid within the next 6 weeks. Any chance?

mikcatta, 2020-04-02:
> To be honest I don't think the fix is going to be anytime soon. 

I really hoped that PiSupply would be more helpful to a fellow UK company, just trying to eek out a living. I have no choice, but to look at alternates now.

tvoverbeek, 2020-04-02:
@mikcatta There is a way you can minimize the 30 sec window.
Instead of using SYS_FUNC_HALT_POW_OFF in case of 'no power' connect the 'no power' event to a user function where you command the shutdown yourself.
This function should also set the timeout (pj.power.SetPowerOff(delay-in-seconds)) just long enough to perform a complete shutdown. Do not enable the 'Software Halt Power Off' on the system task tab, because it does not allow timeouts less than 20 seconds which is most likely longer than what you need for your application.
This way the window for reappearance of power before the PiJuice removes the 5V is very small.

mikcatta, 2020-04-02:
@tvoverbeek Ok, I will give that a go over the next couple of days, thank you :)

mikcatta, 2020-04-12:
It appears from an email that @shawaj /PiSupply are not intending to "fix" the bugs in the firmware anytime soon, so have made the difficult decision that we won't be using the PiJuice product going forward. Many thanks to all those that contributed to this thread.

closed 2020-04-12 by mikcatta