PiSupply/PiJuice #482: GetBatteryCurrent() doesn't match GetIoCurrent() when there is no external power

nowherenone, 2020-03-26:
In my opinion when rPi is on battery power, battery_current should equal io_current with a negative sign.  Please tell me if it is expected, or is there any way to fix the battery profile?
[image]
Currently I'm using Pi Zero + PiJiuce Zero with 12000mAh Battery and PJLIPO_12000 profile.   I also tried Pi3B+ PiJiuce HAT with a stock battery, and I had the same issue in both cases

tvoverbeek, 2020-03-26:
Battery current is estimated by measuring the change in charge level normalized on the battery capacity. There is no direct current measurement.
Same goes for the IO current which is estimated from the state of the DC/DC converter  and/or the LDO regulator taking the temperature dependence of the regulator in account.
That is what I could deduce from the PiJuice schematic.
So one is definitely not the negative of the other.
Maybe @mmilann (the board designer and firmware author) can comment on this

nowherenone, 2020-03-26:
Thank you so much for the clarification!  That explains why battery charge curve looks similar to the battery current one. :)  If there is no direct current measurement on the battery I should just use an external sensor, thanks again!

closed 2020-03-26 by nowherenone