PiSupply/PiJuice #484: cannot change i2c Address2

MaResailing, 2020-03-31:
Hello
First of all, I'm a beginner in this business.
I build a navigation computer on a Raspi4 with openplotter and the Moitessier Hat and the PiJuice Hat.
As soon as I put the moitessier-hat on the PiJuice, I lose the connection to the PiJuice. The moitessier-hat continues to run. With my modest knowledge it looks like a conflict with the double i2c assignment:
only PiJuice:
pi@openplotter:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --      

only moitessier hat:
pi@openplotter:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77   

According to the documentation you should be able to change the value of the I2C Address2 (RTC) in the PiJuice settings under Configure HAT. But nothing happens, no matter what I enter in these fields.
Here is my Raspi environment:
pi@openplotter:~ $ hostnamectl
   Static hostname: openplotter
         Icon name: computer
        Machine ID: 90007509bb7e479ea929fda793e08c79
           Boot ID: 2e313ef7e76840bdab8051a6d394ea84
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 4.19.97-v7l+
      Architecture: arm

Thanks for the help!

tvoverbeek, 2020-03-31:
I assume you have set the2  ID-EEPROM address of the PiJuice to 0x52.
Since there are 2 HATs each with their ID-EEPROM at 0x50 there are conflicts at boot.

I understand the moitessier hat uses I2C addresses 0x68 for the motion tracking sensor and 0x77 for the barometric pressure sensor. It seems you do not have the humidity and temperature sensor which uses address 0x40.

With only the PiJuice HAT connected select the ID-EEPROM address as 0x52 via the dropdown.
Set the I2C Address2 (RC) to something else, e.g. 69. If you are using the CLI  do not forget to <Apply settings>.
Then run `i2cdetect -y 1` again which should show only addresses 0x14 and 0x69.

MaResailing, 2020-04-01:
Thank you for your answer. Yes, I set the EEPROM ID to 0x52. However, the value I entered in the CLI (e.g. 69) is not saved and I always have the same old values after a reboot. Is there any other way than changing the I2C address via the CLI?

tvoverbeek, 2020-04-01:
After you change the RTC address to 69, go down to "Apply settings" and press Enter:
[image]

You should get the "Settings successfullly updated" message:
[image]

After exiting the CLI `i2cdetect -y 1` should show addresses 14 and 69. At least it does for me.

Which version of the software are you running? Look at the output from `dpkg -l pijuice-base`.
Current version is 1.6.

MaResailing, 2020-04-01:
I have the latest version: 
||/ Name           Version      Architecture Description
+++-==============-============-============-=========================================
ii  pijuice-base   1.6          all          Basic support for Pi-Supply's PiJuice HAT

ii  pijuice-base                         1.6                                 all          Basic support for Pi-Supply's PiJu
ii  pijuice-gui                          1.6                                 all          The desktop applications for the P

But I have a different CLI interface than in your screenshot:

[image]

Under Configure HAT, according to the text in the lower right corner, changes should be applied immediately. The Apply button is only available on the CLI homepage

tvoverbeek, 2020-04-01:
Your screenshot shows the address as 69.
Anyway, when you edit the address make sure you press Enter or mve thecursor to another field.
You should then get a popup. See below:
[image]

MaResailing, 2020-04-01:
My screenshot shows my desired input BEFORE I have closed the window. No, no popup is displayed...

tvoverbeek, 2020-04-01:
What version of the software are you running? Look at the output from `dpkg -l pijuice-base`.
Current version is 1.6.
In the GUI can you check the version of the firmware? Current version is 1.4.

MaResailing, 2020-04-02:
After rebuilding the whole system from scratch (including openplotter debain) the problem seems to be solved - the popup window appears after entering the value and the entered value is saved and there is no connection error. Thanks for all the help!  Stay healthy, everybody!

closed 2020-04-02 by MaResailing

tvoverbeek, 2020-04-02:
Glad you got it working. Good luck with your system.