PiSupply/PiJuice #486: Firmware warning breaks headless pijuice-base apt installation

elaurijssens, 2020-04-02:
I'm using headless installations for PiJuice-based systems. However, the installation stalls because the installation *always* waits for a keypress:

```
$ sudo apt-get install -y pijuice-base
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  pijuice-base
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/100 kB of archives.
After this operation, 357 kB of additional disk space will be used.
Selecting previously unselected package pijuice-base.
(Reading database ... 41067 files and directories currently installed.)
Preparing to unpack .../pijuice-base_1.6_all.deb ...
Unpacking pijuice-base (1.6) ...
Setting up pijuice-base (1.6) ...
Setting ownership and permissions
pijuice.service is not active running, attempting to enable
Removed /etc/systemd/system/pijuice.service.
*************************************************************
Until you update the firmware to V1.4 you still need the
'dtoverlay=i2c-bcm2708' work around in /boot/config.txt.
After the firmware update to V1.4 the 'dtoverlay=i2c-bcm2708'
can be removed from /boot/config.txt
*************************************************************

Press any key to continue
```

tvoverbeek, 2020-04-02:
@elaurijssens The message is only displayed when on a pi4.
If you want an unattended installation of pijuice-base use the following:
`echo 'x' | sudo apt-get install pijuice-base`.
PS. Nederlands?

elaurijssens, 2020-04-02:
Hi Ton,

> @elaurijssens The message is only displayed when on a pi4.
> If you want an unattended installation of pijuice-base use the following:
> `echo 'x' | sudo apt-get install pijuice-base`.

That doesn't work in an Ansible playbook, except when I'd do that as an OS command. And then I could probably achieve the same with setting `DEBIAN_FRONTEND=noninteractive` and adding a f or q to the -y. But that kind of defeats the idea of a stateful deployment. The message is possibly useful, but I don't see the point in waiting for a key as some kind of confirmation, as it's a matter of RTFM.
TBH I would expect the message in the CLI after upgrading the firmware to 1.4 or later, not at install time. The installer doesn't even check the firmware nor config.txt AFAIK.

> PS. Nederlands?

Tuurlijk.

tvoverbeek, 2020-04-02:
@elaurijssens The prompt is done by a direct read a single character in the postinst script.
Therefore using `DEBIAN_FRONTEND=noninteractive` or using yes-force will probably not work.
I should have used debconf, but having no experience with debconf I took the easy way.

Sorry dat het niet zonder meer werkt met Ansible.

shawaj, 2020-10-24:
@tvoverbeek is there anything we need to do here? 

Perhaps I could add your comments somewhere in readme?
> @elaurijssens The message is only displayed when on a pi4.
> If you want an unattended installation of pijuice-base use the following:
> `echo 'x' | sudo apt-get install pijuice-base`.
> PS. Nederlands?

elaurijssens, 2020-10-24:
IMHO, as a quick solution, the wait for a keypress should be removed. As a better solution, the message should be moved to the readme. There won't be many older PiJuices with firmware<1.4 that are moved to a RPi4; if people do, habitually copy configs and not upgrade firmware, things will continue to work, will they?

tvoverbeek, 2020-10-24:
@shawaj @elaurijssens Suggest to keep the issue open.
What should be done is to move this to the pre-configure phase, so this will work with software like ansible.
Anyway, with the next release the Pi4 question will be removed.