PiSupply/PiJuice #487: rtc sync problem

angryfrenchguy, 2020-04-03:
I have a little cronjob script on my pi that checks if I'm home (pings my phone), does something (or not), programs an alarm on pijuice (15 minutes if I'm home, longer if I'm not) and shuts down.

Investigating some weirdness, I noticed this in my logs : 

```
2020-04-03 12:56:28,540 | le_log | INFO | *******************************************|
2020-04-03 12:58:22,761 | le_log | INFO | Loggeur opérationel
2020-04-03 12:58:22,771 | le_log | INFO | PiJuice opérationel
2020-04-03 12:58:22,778 | le_log | INFO | L'horloge est online
2020-04-03 12:58:22,781 | le_log | INFO | PiJuice: 2020-4-3  19:11:55 UTC
2020-04-03 12:58:22,783 | le_log | INFO | Pi     : 2020-04-03 12:58:22.783351
2020-04-03 12:58:22,786 | le_log | INFO | Let's go.
2020-04-03 15:11:57,715 | le_log | INFO | ping --> 0
2020-04-03 15:11:57,719 | le_log | INFO | Tu est à la maison
2020-04-03 15:11:57,723 | le_log | INFO | Le Pi n'arrive pas à se brancher à internet. On essaye plus tard.
2020-04-03 15:11:57,943 | le_log | INFO | À tantôt
2020-04-03 15:11:57,953 | le_log | INFO | {'data': {'second': 0, 'minute': 13, 'hour': 'EVERY_HOUR', 'day': 'EVERY_DAY'}, 'error': 'NO_ERROR'}
2020-04-03 15:11:58,305 | le_log | INFO | fini.
2020-04-03 15:11:58,308 | le_log | INFO | *******************************************|
```

Notice how the timestamp is only updated halfway through the code?  I added some check at the begining of my code (lines 5 and 6) to see the discrepancy between the pi and the pi juice.  (Pijuice is UTC. PI timestamp is local, but that is not the problem. You can still see from the minutes that they're not synced. PIjuice time is correct. Local time should be 4 hours behind UTC). 

This causes the time variable I use to be set on the unupdated time. If the code runs long, it could set an alarm in the past.

**What I tried  :**

As per pijuice documentation, I added `sudo hwclock -s` to `/etc/rc.local`. As you can see in the log, it didn't fix it. 

Not sure if it's relevant, but I also do this at the begining of my code :

```
pjOK=False
while pjOK == False:
   stat=pj.status.GetStatus()
   if stat['error'] == 'NO_ERROR':
      pjOK = True
      log.info('PiJuice opérationel')
   else:
      sleep(0.1)

while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)
```

(Copied from some coee on this github I used a my template. Not sure it helps. Doesn't seem to break anything)

EEPROM on Ox52 (stacking 2nd HAT) and `dtoverlay=i2c-rtc,ds1339` in /boot/config.txt

Should I just get my time variable from `rtc_time = pj.rtcAlarm.GetTime()` ? Would fix the problem, but my logs would still be messed up...

tvoverbeek, 2020-04-04:
Whwn the Pi starts the time starts at the time of the previous shutdown. This done by the fakehwclock to ensure an always increasing time. When you are connected to the internet the first run of the timesyncd will synchronise the system time and copy it to the RTC. This is what you see in your log at 15:11:57,715
If you run `sudo hwclock -s` from ./etc/rc.local it runs very early in the boot sequence even before i2c is initialised. Hence it fails.
Run it when you get a 'NO_ERROR' status from pj.status.GetStatus(). This also set the system time when there is no internet connection.

Normal PCs have a built-in RTC which is available immediately at boot (no i2c communication), then it works fine when run from /etc/rc.local

angryfrenchguy, 2020-04-05:
Perfect. Seems to have fixed it. Many thanks. Brilliant little gadget.

closed 2020-04-05 by angryfrenchguy