PiSupply/PiJuice #488: PiJuice technical data

fsegb, 2020-04-04:
Hello,
I'm designing a system that will be solar powered. I'm interested in this product but I need to know the following information:
1. Nominal voltage
2. Max power intensity
3. Short-circuit intensity
4. Functionality factor

I'm interested in the 40W model but I would appreciate receiving the information for every model.
Do you have a technical data sheet?

Thanks in advance.

shawaj, 2020-04-04:
The PiJuice Solar panels have a regulated 5v output. The 40 watt panel also has a regulated 19 volt output via the DC jack.

Are you planning to use this panel in an implementation without the PiJuice HAT itself?

shawaj, 2020-10-24:
@fsegb do you need anything else here or can I close the issue?

fsegb, 2020-10-26:
You can close it, thank you very much
Francisco

closed 2020-10-26 by shawaj