PiSupply/PiJuice #489: Can I set a wakeup alarm several days in the future?

angryfrenchguy, 2020-04-05:
I can *input* year and month variables with `pj.rtcAlarm.SetAlarm(x)`.

`pj.rtcAlarm.GetTime()` *returns* years and months : 

`{'data': {'second': 5, 'minute': 34, 'hour': 18, 'weekday': 2, 'day': 31, 'month': 3, 'year': 2020, 'subsecond': 0, 'daylightsaving': 'NONE', 'storeoperation': False}, 'error': 'NO_ERROR'}` 

`pj.rtcAlarm.GetAlarm()`, however, does not return years and months : 

`{'data': {'second': 0, 'minute': 48, 'hour': 'EVERY_HOUR', 'day': 'EVERY_DAY'}, 'error': 'NO_ERROR'}`

Apparently I can do this on the GUI [Set wakeup alarm for long durations · Issue #439 · PiSupply/PiJuice · GitHub](https://github.com/PiSupply/PiJuice/issues/439)

*CAN I* set a wakeup alarm days, months, *years?* in the future with python? So far I haven't been able to and have had to program a daily checkin.

tvoverbeek, 2020-04-05:
You can specify keys 'year' and 'month' to `pj.rtcAlarm.SetAlarm()` but they are ignored.
In Issue #439 there is no example of year and month.

What you can do is either specifying 'weekday' (1=Monday, 2=Tuesday, ...) as a single day or a ';' separated list of several weekdays ('1;3' is Monday and Wednesday) or specifying 'day' as a day in the month.

angryfrenchguy, 2020-04-06:
Ok. Thanks!

quitmeyer, 2024-03-12:
> Can one specific multiple days in a month, or does it always throw an error?

ahhh it appears in the software, there is no checking for multiple dates in a month

```
        if 'weekday' in alarm:
            try:
                day = alarm['weekday']
                if day == 'EVERY_DAY':
                    d[3] = 0x80 | 0x40

                elif isinstance(day, str) and day.find(';') < 0:
                    dw = int(day)
                    d[3] = d[3] | (dw & 0x0F) | 0x40

                elif isinstance(day, int):
                    dw = int(day)
                    d[3] = d[3] | (dw & 0x0F) | 0x40

                elif (isinstance(day, str) and day.find(';') >= 0):
                    d[3] = 0x40 | 0x80
                    ds = 0x00
                    dl = day.split(';')
                    dl = dl[0:-1] if (not bool(dl[-1].strip())) else dl
                    for i in dl:
                        ds = ds | (0x01 << int(i))
                    d[8] = ds
            except:
                return {'error': 'INVALID_WEEKDAY'}
        elif 'day' in alarm:
            try:
                day = alarm['day']
                if day == 'EVERY_DAY':
                    d[3] = 0x80

                else:
                    dm = int(day)
                    d[3] = (((dm // 10) & 0x03) << 4)
                    d[3] = d[3] | ((dm % 10) & 0x0F)
            except:
                return {'error': 'INVALID_DAY_OF_MONTH'}
        else:
            d[3] = 0x80  # every day

```

is it possible to somehow pass a hex value or something directly to overcome this @tvoverbeek ?