PiSupply/PiJuice #490: user_func issues

slack17, 2020-04-07:
I've installed the latest software of GUI and CLI, 
and I'm not able to get the user_func1 to work, 

I've defined the user script in both GUI and CLI both are not working,
the chmod is fine of the file, file defined correctly with header of the bash script or python, but still no action from user_func script, 
I setup the low % when it goes below it should run the script and low battery voltages, still same issue. 
also I've restarted the device couple of times. . 

could you let me know what it could be the issue ? I'm using latest raspbian buster.
Thanks

tvoverbeek, 2020-04-07:
The actual script has to be written outside the GUI/CLI. The script can be a python/bash/perl/... script and has to be executable and owned by a user belonging to the pijuice group. scripts owned by root will not be executed (security reasons).
Can you give us more details from your setup? I understand you have connected USERFUNC_1 to low battery. On the user functions tab you have to tell that USER_FUNC_1 is e.g. /home/pi/my_function.py
If possible include also a listing of your script.

slack17, 2020-04-07:
Dear Mr Ton,
thank you and appreciate your quick support. 

first step I've defined both low battery ( checked )
and low battery as voltage also checked, 
and I make sure on the stop task is enabled to run all script on the below cases. 

on the next screen where I define the user script. I put the full path of the script.
I've tried both way python /home/pi/myscript.py  & tried with /home/pi/myscript.py 
+ on the top of the script file I've added. #!/usr/bin/python also same for bash script .sh

I've did the chmod correctly for the file to be executed as mention 
sudo chmod +x file.py 

the last thing I copied one of your test script in the folder "test" user_func1. 
manually I can run the script it will create a .log file but in the pijuice system nothing is calling the scripts so far. 

restarted the system couple of times. tried in CLI and GUI to enable and define the scripts same with nothing is running. 

anything to be added in the crontab ? or it should automatically should check based on the services running ?

Thank you,

tvoverbeek, 2020-04-09:
No need to add anything to crontab.
But it is essential the pijuice service is running. ("system task enable" in the GUI or CLI)
Check in a terminal with `systemctl status pijuice.service`.

slack17, 2020-04-09:
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-04-09 01:42:48 +03; 13h ago
 Main PID: 364 (pijuice_sys.py)
   Memory: 9.5M
   CGroup: /system.slice/pijuice.service
           └─364 /usr/bin/python3 /usr/bin/pijuice_sys.py

I've the latest Raspbian 3.3.1 
latest software 1.6 Firmware 1.4 , 
I've install it again in different Rpi still with the same issue. 

-even when I put 90% to wake up when it reaches this percentage it doesn't work. 

I guess only one it was working fine with pre-loaded functionalities if I pick one it will work, I will try that again. with shutdown function.
but when I assign my functions it doesn't work.

tvoverbeek, 2020-04-09:
Your own function needs to be executable.
E.g. if it is a python script the first line has to be `#!/usr/bin/python3`.
The function file needs to be executable, if not `chmod +x <yourfunction.py>`
The owner of the script needs to be a member of the pijuice group.

slack17, 2020-04-09:
Yes all 3 points are correct and I doubled checked,
first line of the script, ( since I'm using your test script in test folder called user_func1.py ) 
when I run as pi user ./user_func1.py then I can see the output of log file in same folder. 
- chmod already there and I did it again on the script. 
- pi user already added into the pijuice groul and I've double checked on it,

strange thing is default pre-loaded scripts withing the pijuice it's running fine. 
for example if I choose SYS_FUNC_SYS_OFF_HALT, 

it's working fine, but when I define my script it doesn't work. !
should I change the folder of the script or anything else I should test ?

Thank you,

tvoverbeek, 2020-04-09:
In the user functions tab is USER_FUNC1 defined as /home/pi/user_func1.py?
If so, copy the user_func1.py from the test directory to /home/pi or change the definition to the path of the test directory.

slack17, 2020-04-09:
In the user functions tab is USER_FUNC1 defined as /home/pi/user_func1.py?
Yes, defined correctly and I tried both ways with /home/pi/user_func1.py And **python** /home/pi/user_func1.py = still same issue. 

this script already in /home/pi/
my question if I can copy this script to /pijuice system files ? I don't know if it's possible or where it's located. I found only firmware file folder and images. 
I thought the predefined scripts already in specific folder within the system ? 

anything else to check ?

tvoverbeek, 2020-04-09:
If everything is as you say it should work.
Can you post the output of `ls -l /home/pi`.
Your version number 3.3.1 does not exist for Raspbian (it could be from R, the statistical language).
Also post the output of `uname -a` and `lsb_release -a`.

slack17, 2020-04-09:
sorry I mean latest NOOBS version 3.3.1 - downloaded and tried this in 2 different pi zero, same issue, here is the output below.

Here is **ls -l** 
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Desktop
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Documents
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Downloads
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:03 MagPi
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Music
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Pictures
-rw-r--r-- 1 root root  194 Apr  8 00:18 pijuice2.py
-rwxrwxrwx 1 root root  987 Apr  8 00:16 pi_new.py
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Public
drwxr-xr-x 2 pi   pi   4096 Apr  8 00:16 __pycache__
drwxr-xr-x 2 pi   pi   4096 Feb 13 19:32 Templates
-rw-r--r-- 1 pi   pi     49 Apr  9 18:19 user_func1.log
-rwxr-xr-x 1 root root  325 Apr  9 17:48 user_func1.py

**uname -a**
Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux

**lsb_release -a**
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

slack17, 2020-04-09:
yes, 
I've deleted the files and I created the files again without sudo command
2020-04-09 20:44:40,320 INFO ['./user1.py']
2020-04-09 20:45:14,729 INFO ['./user1.py']
2020-04-09 20:51:58,526 INFO ['/home/pi/user1.py', 'low_charge', '93.0']
2020-04-09 20:52:04,787 INFO ['/home/pi/user1.py', 'low_charge', '93.0']
2020-04-09 20:52:11,021 INFO ['/home/pi/user1.py', 'low_charge', '93.0']

appreciate your support.

it's working fine. I'm just testing the wake up task based on % charge level. last time it wasn't functional. 
I've just turn it off and I put the charge on. when it reaches the specific level % or above it will turn on ?

slack17, 2020-04-22:
Dear Ton thank you for your support and I hope you doing well,

now I'm able to run the specific script, but I still can't get the Rpi device to turn ON when it reaches xx% of the charge. anything from the system I can check for this ?
I need to test this feature before I put it outside in the field.

tvoverbeek, 2020-04-22:
@slack17 It depends on how you shut down the Pi before you want to wake it up.
If you use a user script to do the shutdown then you have to enable the wakeup-on-charge flag in the script. When using SYS_FUNC_HALT_POW_OFF the service will do this for you using the setting specified in the GUI/CLI.

mike9665, 2020-05-22:
Hi
I am also unable to execute a user function script for any of the pijuice system events. I have confirmed the pijuice group for user pi, the script are r-x and owned by pi.

I'm running Buster and the pijuice is at s/w 1.6 and f/w 1.4 is there anything else I can check or do to help the situation. Incidentally I can get the RPi to reboot if I change the system event to SYS FUNC REBOOT.
Mike

mike9665, 2020-05-22:
Hi
I have resolved the issue. It was caused by dos cr/lf characters in the shebang. 

Mike

tvoverbeek, 2020-05-22:
@mike9665 Ok to close the issue then? If you do not agree, please reopen it.

closed 2020-05-22 by tvoverbeek