PiSupply/PiJuice #492: How to activate "wake-up on charge" and "watchdog" for normal shutdowns?

lg007, 2020-04-11:
Hello,

I bought 2 pijuice modules and find them very versatile and useful. I have to deploy both of them in very remote places where human interaction will not be possible. Both will have mobile Internet connections with some limited upgrade and access possibilities. That is why I need to make both of them almost impossible to stop or switch off. For this I would like to have the "wake-up on charge" and "watchdog" functionalities to work even if a user (or program) issues a "shut down" command or long presses the s1 button (right now after issuing a "shutdown" command the raspberry stops, power is switched off and nothing happens later, even if charging power comes back - the only way to wake up the raspberry is by pushing the s1 button). I know this should be possible by modifying the pijuice service code. Instead of reverse engineering the python code, I am asking here what should be done to make this possible. I hope it is not too much to ask!   

Thanks,

NB.: What I also miss is the possibility of run a user script when the **power comes back** and the raspberry is still working on UPS... (not a script when the raspberry starts, but when power - and charge - is available again working under UPS power)

tvoverbeek, 2020-04-13:
At shutdown pijuice_sys.py disables the watchdog. Normally the watchdog is only active when the pijuice service (pijuice_sys.py)is running and the watchdog is enabled.
If you want the watchdog to reboot the Pi after shutdown comment out the disabling of the watchdog (lines 289 - 300 of pijuice_sys.py) and restart the service.
Note: Pi will restart after every shutdown until the battery is empty.

When using the "Software Halt Power Off" option the wakeup-on-charge setting is not copied to the PiJuice at shutdown.
If you want this add the line `pijuice.power.SetWakeUpOnCharge(0)` after line 314
(`pijuice.power.SetPowerOff(powerOffDelay)`) in pijuice_sys.py.

lg007, 2020-04-13:
Thanks, I will try your suggestions.

closed 2020-04-13 by lg007