PiSupply/PiJuice #493: Power Off (5V) not switching off when running on battery

olehj, 2020-04-12:
I have a 12000mAh battery attached to a PiJuice Zero board, and I have set it up to run the way I want, except from one part:
When the solar panel is covered/not getting any light and does not generate power anymore, the Pi will not shut off the 5V leaving the 3G USB modem connected via OTG, and probably leave the power on for other devices as well. But when the solar panel is on and generating power and charging the battery, everything works as expected, and the Pi is powering on, running the program and shutting down every hour.

The Pi does still "wake up" every hour to do the program job, but simply seem to ignore the power off function for the PiJuice. It does however seem to turn off the RasPi, just not cutting the 5V power to it.

Any setting I have missed here? The Pi Zero 1.3 (non-W) is running full read-only and rely on the RTC clock and settings from the PiJuice only. I would like to shut off the 5V entirely when th Pi is off, as the 3G modem uses unnecessary power and will run 24/7 with ~3 minutes runtime every hour.

tvoverbeek, 2020-04-13:
How do you shut down the Pi after the 3 minutes run time?
How do you know the 5V is off when the solar panel is providing power?
Have you enabled the "Software Halt Power Off" setting on the "System Task" tab?

olehj, 2020-04-13:
This is the shutdown script used, which works most of the time:
`import os, pijuice, time, subprocess

while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

pj = pijuice.PiJuice(1, 0x14)

pj.power.SetPowerOff(30)
subprocess.call(["sudo", "poweroff"])`

I think the "System Tasks" are mainly non configured, as I thought the shutdown script above would deal with the task I require it to do.

olehj, 2020-04-13:
I have enabled the SW Halt Power Off, and will let it run during the night and further out next day and see how it behaves. I'll keep you updated.

olehj, 2020-04-14:
Looks like it is working now with the System Task Halt enabled! TY

closed 2020-04-14 by olehj