PiSupply/PiJuice #495: Repeat wakeup not working when on battery power

hystrix1, 2020-04-22:
I have tried for days to get repeat alarms to work when on battery power. If I power the PiJuice from the microUSB connector, wakeup works every time. 
However, as soon as I remove power to the PiJuice, the next wakeup works OK, then on subsequent restarts it seems the RTC no longer keeps the correct time (and eventually sets a wakeup time in the past).
Is there a definitive solution to get this working? I have been through the previous reported issues regarding the RTC (e.g https://github.com/PiSupply/PiJuice/issues/79 and seem to have tried everything (e.g. EEPROM = 52, dtoverlay=i2c-rtc,ds1339).

Here's my configuration:
PiJuice HAT v2.2.2 with BP7X battery
v1.4 firmware
PiZeroW, connected to internet via WiFi. Camera module attached. (OS = Buster, fake hwclock disabled)
EEPROM = 50
dtoverlay=i2c-rtc,ds1307 added to /boot/config.txt (although I understand this is not actually required) I get 'UU' in i2cdetect -y 1 for 68
No pogo pin

wakeup_enable.py run from cron @reboot with the following:

```
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

PiJuice_OK = False
pj = pijuice.PiJuice(1, 0x14)
# wait until we can read from PiJuice
while PiJuice_OK == False:
    try:
        current = float(pj.status.GetIoCurrent()['data'])/1000
        PiJuice_OK = True
    except KeyError:
        pass
```

Script to set wakeup time also run from cron @reboot, with the following:

```
while not os.path.exists('/dev/i2c-1'):
    time.sleep(0.1)

# access to pijuice and wait for valid data
PiJuice_OK = False
pj = pijuice.PiJuice(1, 0x14)
while PiJuice_OK == False:
    try:
        current = float(pj.status.GetIoCurrent()['data'])/1000
        PiJuice_OK = True
    except KeyError:
        pass
        
# Rely on RTC to keep the time
cmd = "sudo hwclock --hctosys"
os.system(cmd)

DELTA_MIN = 10
d_wake = dict()
d_wake['year'] = 'EVERY_YEAR'
d_wake['month'] = 'EVERY_MONTH'
d_wake['day'] = 'EVERY_DAY'
d_wake['hour'] = 'EVERY_HOUR'
t = dt.datetime.utcnow()
d_wake['minute'] = (t.minute + DELTA_MIN) % 60
d_wake['second'] = 0

status = pj.rtcAlarm.SetAlarm(d_wake)
if status['error'] != 'NO_ERROR':
    print('Cannot set alarm\n')
    sys.exit()
else:
    print('Alarm set for ' + str(pj.rtcAlarm.GetAlarm()) + '\n')
    
time.sleep(180)

# Enable wakeup, otherwise power to the RPi will not be
# applied when the RTC alarm goes off
pj.rtcAlarm.SetWakeupEnabled(True)

# clear alarm flag
pj.rtcAlarm.ClearAlarmFlag()

# PiJuice shuts down power to Rpi after 30 sec from now
# This leaves sufficient time to execute the shutdown sequence

pj.power.SetPowerOff(30)
subprocess.call(["sudo", "poweroff"])
```

Everything works until power is removed from the PiJuice and it is running on battery power.

My aim is to have the PiJuice solar powered and to power on at sunrise and power off at sunset (trivial to program), every day.

tvoverbeek, 2020-04-22:
Have you tried to test how long you can run the Pi on battery power without shutting down?
It could be your BP7X is not healthy and only allows a single reboot before it cannot deliver the needed power any more. My own BP7X batteries are dead, seem to charge OK, but as soon as you put a load on them they go down.

hystrix1, 2020-04-23:
I checked the battery - 97% to 84% in 1 hour, so that's OK, but from your comment, it's something I will keep an eye on.

I added a while loop to my code to wait until the System time has synchronised with the RTC time, within +/-1 second. It looks like the System time can take 20 seconds or more before it comes in line with the RTC time. Obviously setting the wakeup time from the Pi time before it was the same time as the RTC was where the problem started.

```
Pi time: 2020-04-23 11:09:45.431965
RTC time: 2020-04-23 11:17:00.000000
Time difference between Pi and RTC: -434.568035 seconds
Pi time: 2020-04-23 11:09:46.440804
RTC time: 2020-04-23 11:17:01.000000
Time difference between Pi and RTC: -434.559196 seconds
Pi time: 2020-04-23 11:09:47.451205
RTC time: 2020-04-23 11:17:02.000000
Time difference between Pi and RTC: -434.548795 seconds
Pi time: 2020-04-23 11:17:02.844505
RTC time: 2020-04-23 11:17:03.000000
Time difference between Pi and RTC: -0.155495 seconds
```
I'm not sure why `sudo hwclock --hctosys` is not working, but I have a solution and my Pi has been waking and sleeping every 10 minutes for nearly 2 hours now, just on battery.

shawaj, 2020-10-24:
@hystrix1 will close this issue now. Feel free to reply if you do not agree

closed 2020-10-24 by shawaj