PiSupply/PiJuice #496: Configuration via file

brettbeeson, 2020-04-22:
I'd like to configure a new Pijuice Zero from scratch using a file. 

Currently I use pijuice_cli to manually setup the unit. I see that /var/lib/pijuice/pijuice_config.JSON contains some settings, as [documented](https://github.com/PiSupply/PiJuice/tree/master/Software#json-configuration-file). But not all: for example, battery setup.

Is there a mechanism to do a 'one shot' automated configuration?

tilghman, 2020-07-01:
This is also something I'm looking for.  I'm looking at being able to mass-load settings on a fleet of devices, instead of needing to load each setting individually.

Do I need to dissect pijuice_cli and create a tool of my own, or is this something that can already be done?

tilghman, 2020-07-06:
I just created a pull request.  I believe that this issue might be duplicative of issue #88.

shawaj, 2020-11-14:
@brettbeeson could you take a look at #518 and see if that utility meets your needs?

brettbeeson, 2020-11-15:
I'll need to buy another hat to test, which I've been meaning to do. I'd like to buy a few and combine shipping. crowdsupply seems cheapest (to Australia). Can you recommend a different source for multiple units and combined postage? Cheers.

shawaj, 2020-11-15:
They do have them in stock at Core Electronics in Australia.

PiJuice HAT - https://core-electronics.com.au/pijuice-hat.html

Or PiJuice Zero - https://core-electronics.com.au/pijuice-zero-a-portable-power-platform-for-raspberry-pi-zero.html

Alternatively Pi Supply will have stock in about 2 weeks and we can ship direct to Australia but maybe not the cheapest option.

shawaj, 2020-11-15:
Sorry forgot to tag you @brettbeeson

shawaj, 2020-12-05:
Hey @brettbeeson going to close this issue now as haven't heard back in a while.

If this doesn't work for you then please open a new issue or reply back here and we can reopen this one.

closed 2020-12-05 by shawaj

brettbeeson, 2021-01-14:
Hi @shawaj I finally bought another pijuice and tested this. It works fine. Thanks! 

ps. The --load and --dump arguments are a bit confusing in command line output (i.e. lines 16, 17)

shawaj, 2021-01-14:
> ps. The --load and --dump arguments are a bit confusing in command line output (i.e. lines 16, 17)

@brettbeeson can you elaborate a bit on what you mean by this? and how to reproduce what you are seeing?

brettbeeson, 2021-01-31:
Just that "all settings" is not informative. Perhaps "write all settings in JSON format to stdout for later --load'ing" or similiar.
```
 g.add_argument('--dump', action='store_true', help='print(all settings)')
 g.add_argument('--load', action='store_true', help='print(all settings)')
```

tvoverbeek, 2021-02-01:
@brettbeeson Tries to improve the help texts and some other fixes.
See commit https://github.com/PiSupply/PiJuice/commit/3b9369502b33cbbb3a79a1e191fb4c008abd7cfc