PiSupply/PiJuice #497: Switch to Turn off Battery Power to PiJuice

rlsten, 2020-04-29:
I have been working on a project to use a Raspberry Pi 4, a PiJuice, and a PiJuice battery in a 3D-Printed enclosure as a portable music source for a USB DAC. I understand that the PiJuice uses about 50-150 mAh of power even when the system is powered down. While this is not a lot of current, it nevertheless will drain the battery.

To maximize battery life, I would like to insert a slide switch between the battery and PiJuice that I would turn on when I want to run the Raspberry Pi and turn off after I've powered down the RPi. I would also have a second momentary switch attached to the J5 port on the PiJuice to turn on the Raspberry Pi after I've sent battery power to the PiJuice using the slide switch.

Do you think this setup would harm the PiJuice or cause any other problems?

Thanks for your help!

Rod

tvoverbeek, 2020-04-30:
This would work, but every time to shut off power to the PiJuice the PiJuice microcontroller goes off -> No more Real TIme Clock. Things like (recurring) alarms to wake up the Pi will not work.
Every time you apply power to the PiJuice its MCU will boot from scratch. Only its non volatile parameters will be restored.

rlsten, 2020-04-30:
Thanks very much! 
Are the settings for the battery (size, charging rate, etc) non volatile?

tvoverbeek, 2020-04-30:
Yes, The battery profile parameters are non-volatile.

rlsten, 2020-04-30:
Thanks again!

closed 2020-04-30 by rlsten