PiSupply/PiJuice #498: Build on Ubuntu 20.04 with Python 3.8 failed

MarkusLange, 2020-05-05:
Hello,

I get serveral error Messages when I try to compile it on Ubuntu 20.04 with Python 3.8

> ubuntu@ubuntu:~/PiJuice/Software/Source$ cd Setuid-Wrapper
> ubuntu@ubuntu:~/PiJuice/Software/Source/Setuid-Wrapper$ sudo cp p* $HOME/PiJuice/Software/Source/bin
> ubuntu@ubuntu:~/PiJuice/Software/Source/Setuid-Wrapper$ cd ..
> ubuntu@ubuntu:~/PiJuice/Software/Source$ sudo ./pckg-pijuice.sh
> running bdist_deb
> running sdist_dsc
> running sdist
> running check
> warning: sdist: manifest template 'MANIFEST.in' does not exist (using default file list)
> 
> warning: sdist: standard file not found: should have one of README, README.txt, README.rst
> 
> writing manifest file 'MANIFEST'
> creating pijuice-base-1.6
> creating pijuice-base-1.6/bin
> creating pijuice-base-1.6/data
> creating pijuice-base-1.6/data/firmware
> creating pijuice-base-1.6/src
> making hard links in pijuice-base-1.6...
> hard linking pijuice.py -> pijuice-base-1.6
> hard linking setup.py -> pijuice-base-1.6
> hard linking bin/pijuice_cli -> pijuice-base-1.6/bin
> hard linking bin/pijuiceboot -> pijuice-base-1.6/bin
> hard linking data/020_pijuice-nopasswd -> pijuice-base-1.6/data
> hard linking data/firmware/PiJuice-V1.4_2020_01_17.elf.binary -> pijuice-base-1.6/data/firmware
> hard linking src/pijuice_cli.py -> pijuice-base-1.6/src
> hard linking src/pijuice_sys.py -> pijuice-base-1.6/src
> Creating tar archive
> removing 'pijuice-base-1.6' (and everything under it)
> dpkg-buildpackage: info: source package pijuice-base
> dpkg-buildpackage: info: source version 1.6-1
> dpkg-buildpackage: info: source distribution unstable
> dpkg-buildpackage: info: source changed by Ton van Overbeek <tvoverbeek@gmail.com>
>  dpkg-source --before-build .
> dpkg-checkbuilddeps: error: Unmet build dependencies: dh-python
> dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
> dpkg-buildpackage: warning: (Use -d flag to override.)
> Traceback (most recent call last):
>   File "setup.py", line 55, in <module>
>     setup(
>   File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
>     dist.run_commands()
>   File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
>     self.run_command(cmd)
>   File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
>     cmd_obj.run()
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/command/bdist_deb.py", line 33, in run
>     self.run_command('sdist_dsc')
>   File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
>     self.distribution.run_command(command)
>   File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
>     cmd_obj.run()
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/command/sdist_dsc.py", line 136, in run
>     build_dsc(debinfo,
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/util.py", line 1501, in build_dsc
>     dpkg_buildpackage('-S','-sa',*args,cwd=fullpath_repackaged_dirname)
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/util.py", line 562, in dpkg_buildpackage
>     process_command(args, cwd=cwd)
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/util.py", line 184, in process_command
>     check_call(args, cwd=cwd)
>   File "/usr/local/lib/python3.8/dist-packages/stdeb3/util.py", line 38, in check_call
>     raise CalledProcessError(retcode)
> stdeb3.util.CalledProcessError: 3
> dpkg-buildpackage: warning: using a gain-root-command while being root
> dpkg-buildpackage: info: source package pijuice-base
> dpkg-buildpackage: info: source version 1.6
[... 122 lines cut ...]
Using this skript:
```
#!/bin/bash

sudo apt update
sudo apt install -y python3-smbus python3-pip python3-stdeb dh-systemd debhelper fakeroot python3-all
sudo pip3 install stem stdeb3

sudo git clone https://github.com/PiSupply/PiJuice.git

cd $HOME/PiJuice/Software/Source/Setuid-Wrapper
sudo ./build-setuid-progs.sh
sudo cp p* $HOME/PiJuice/Software/Source/bin

cd $HOME/PiJuice/Software/Source/
sudo ./pckg-pijuice.sh

sudo cp $HOME/PiJuice/Software/Source/deb_dist_base/pi*.deb $HOME
sudo cp $HOME/PiJuice/Software/Source/deb_dist_gui/pi*.deb $HOME

sudo rm -rf PiJuice

sudo apt install -y python3-urwid lua5.3
```
Need help

MarkusLange, 2020-05-05:
Can correct myself dh-python was missing:
`sudo apt install dh-python`
solv it
but in line 116 from pijuice_cli.py is since Python 3.8 an Syntax error

> /usr/bin/pijuice_cli.py:116: SyntaxWarning: "is not" with a literal. Did you mean "!="?
>   if default is not None and default is not "":

it's gone if you change the 2nd **is not** to **!=** so that you have this line

`116:             if default is not None and default != "":`

after this all runs as smooth as usual

tvoverbeek, 2020-05-05:
@MarkusLange Thanks for the syntax correction for Python 3.8. Updated pijuice_cli.py: commit 
https://github.com/PiSupply/PiJuice/commit/87fe15ad0793f0cb2b3a720e4c0752b98672025c

closed 2020-10-24 by shawaj