PiSupply/PiJuice #499: Under-voltage detected

flobernd, 2020-05-07:
Hi there,

after I got a massive data-corruption I had a look in the logs of my Raspberry Pi 4 and found a lot of recently occured `Under-voltage detected` messages followed by a boot sequence. I can't say for sure if this only happens on high CPU usage/stress.

I was using the PiJuice to exactly prevent this from happening (as the database I'm running on there is well known to get corrupted if it is not shutdown gracefully).

Now it seems like the Pijuice is not able to supply enough power to the Pi. I did not overclock and I don't use any USB device.

When I bought the Pijuice I was already concerned because the power output is less than the recommended 2,5A. I remember asking here if the Pijuice will still be sufficient and you answered with "yes". You were right until now - the setup worked for several months.

Any suggestions?

Best regards

tvoverbeek, 2020-05-07:
The battery will go down in capacity over time. Obviously the capacity is now too low to continuously support your configuration. Are you using an external disk for your database?
If so try to use a separate power supply for the disk to minimize the current the PiJuice has to supply. Also make sure your Pi 4 is up-to-date. There have been some updates to its boot rom which lower its current consumption.

flobernd, 2020-05-08:
Well, I'm not sure if this is related to the battery or some other faulty component. Anyways - this is kinda disappointing after less than one year of use.

Did some tests and if I unplug the power the Pi powers off completely. After 2-3 seconds it turns back on tho (sometimes). It definitely became totally unrelieable (instead it's even causing severe power-loss) and I can't use it for my setup any longer.

The Pi does not have any USB devices plugged in. Currently the database (it's only a small MongoDB with very few writes) is on the SD card (I know it wears out the card - planned to migrate to network boot for some time already). But this can't be the issue.

The firmware of the PiJuice is up-to-date (checked for updates every month or so and just checked again today).

tvoverbeek, 2020-05-09:
> And what's even more interresting: The device runs fine if I only power it with the battery. As soon as I plug in the power-cable (into the PiJuice connector) it immediately turns off.

Turns off abruptly or does it execute a shutdown?

Ir smells like you  eitherhave a screwed up firmware/firmware settings and/or a flaky BP7X.
Have you tried to reset the PiJuice to default values? (Re)updating the firmware does not reset the internal non-volatile parameters.

My own BP7X batteries are dead after using them since the Kickstarter

flobernd, 2020-05-10:
> Turns off abruptly or does it execute a shutdown?

It turns off abruptly.

> Ir smells like you eitherhave a screwed up firmware/firmware settings and/or a flaky BP7X

I did not change any settings since the initial configuration. The only thing I did was a firmware update from time to time. Might indeed be the battery but sadly I don't have a spare one to test. 

In the meantime I connected the Pi to my APC UPS and configured it to serve as a NUT server. Might be the better option for 24/7 use.

tvoverbeek, 2020-05-10:
@flobernd Pi-Supply announced that they are going to have BP7X replacement batteries available in the near future.
See e.g. https://uk.pi-supply.com/products/pijuice-1820mah-replacement-battery-for-pijuice-hat

flobernd, 2020-05-11:
Thanks for the hint, but I don't think I will get me a new battery. It's just not worth for my usecase if it burns out that fast.

tvoverbeek, 2020-05-11:
OK. Good luck with your project.
Closing the issue.

closed 2020-05-11 by tvoverbeek

flobernd, 2020-05-11:
> OK. Good luck with your project.

Thanks! 

Btw not meant to be negative in any way. I liked the PiJuice so far - just not the right thing for 24/7 usage I guess.