PiSupply/PiJuice #500: Negative GPIO power input / current

jophut, 2020-05-14:
I noticed sometimes, the status screen from the pijuice_cli gives a negative current:

I don't understand this because the system is running on battery only.

[image]

Currently it doesn't bother me, but I just wanted to let you know.
Maybe you have a explanation for this situation?
Could it simple be a faulty AD-conversion?

tvoverbeek, 2020-05-14:
@jophut There is no AD conversion. The current is estimated by an algorithm in the PiJuice firmware. I do not know the details, but I've seen this switching regularly on my own PiJuice.
So nothing to worry about,

shawaj, 2020-05-16:
Does it go to positive when you plug in power to the RasPi microUSB?

I guess this means it is drawing 1.206A of power from the PiJuice, rather than a positive number which would mean it is supplying power from the GPIO to the PiJuice.

Just a thought - I am not sure this is the case.

shawaj, 2020-10-24:
@tvoverbeek do you think it's worth documenting this algorithm? Could ask Milan if so

tvoverbeek, 2020-10-24:
Added clarification about GPIO current sign to both GUI and CLI sections of the README. (https://github.com/PiSupply/PiJuice/commit/d3c193c9f40ffc2f45cdb9304cd7c59ee46952b2).
Closing the issue.

closed 2020-10-24 by tvoverbeek