PiSupply/PiJuice #501: PiJuice unable to power Media Center HAT

jsteele-cnn, 2020-05-16:
After waiting 5 years, my Maker Kit arrived the other day.
I decided to put it together today, and followed the instructions.
When the time came to remove the HDMI cable and begin outputting video to the Media Center HAT, nothing happened.  
After retrying around 10 times with various options selected from the script that is supposed to configure the HAT, I finally decided to try without the PiJuice in place, and suddenly the HAT worked.
I tried to use 2 different PiJuices, but the result was the same for both.
The Media Center HAT _does not work_ with the PiJuice in place.
I can't say I'm shocked with this incredible incompetence from a team that is 5 YEARS late delivering a couple of simple products, but I'm certainly pissed, and will make sure to tell everyone I know to avoid this company like the fucking plague.

ChristopherRush, 2020-05-16:
You could just try changing the EEPROM address on the PiJuice to 0x52. I
will ensure this is well documented on GitHub and our user guides.

jsteele-cnn, 2020-05-16:
"well documented"  

You can't be serious.
Show me where: https://learn.pi-supply.com/make/pijuice-games-console-project/

ChristopherRush, 2020-05-16:
I stated `I
will ensure this is well documented on GitHub and our user guides`
I have updated GitHub https://github.com/PiSupply/Media-Center-HAT/commit/64207447ba7743b6130c3c2f4bff8b866cbafd86 and Games console project with the correct URL.

jsteele-cnn, 2020-05-16:
It would certainly be more helpful if you explained HOW to actually DO what's described by "When using the Media Center HAT with the PiJuice HAT you will need to change the ID EEPROM address on the PiJuice to 0x52 otherwise both HAT's will conflict with each other and the Media Center HAT will not load the DT overlay file."

tvoverbeek, 2020-05-16:
@jsteele-cnn Please read the Software README on Github. There the procedure to change the address is documented (https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-general-config-menu).
Since you have several PiJuices I assume you know how to set them up.

ChristopherRush, 2020-05-16:
Agreed, I will write up a user guide on how to do this process. For the time being you can change it here https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-general-config-menu or here https://github.com/PiSupply/PiJuice/tree/master/Software#general. Make sure you check "ID EEPROM Write Unprotect" and then change it from the drop down box.

jsteele-cnn, 2020-05-16:
Still not terribly clear.
When I'm running a regular linux distro, I can install the pijuice gui, and I could modify shit in there, but that's not what this is, this is just the RetroPie image.
Am I supposed to just install the pijuice-base package to do this?
I don't understand why getting a straight answer on this has to be like pulling teeth.

ChristopherRush, 2020-05-16:
Yes, you will need to install the pijuice-base package to change edit any of the PiJuice settings otherwise it is just a dumb UPS. `sudo apt-get install pijuice-base`

jsteele-cnn, 2020-05-16:
ok, so with that installed, when I run the pijuice cli, I need to:
- check the `EEPROM write unprotect` option
- change `EEPROM address: 50` to be `52`
Is that correct?
Is there anything else that I should need to do?

jsteele-cnn, 2020-05-16:
That appears to have worked.
Thank you for the clarification.  
I implore you to clarify these items here so that others who finally received their backer reward don't waste 7 hours of their weekend fighting this thing: https://learn.pi-supply.com/make/pijuice-games-console-project/

ChristopherRush, 2020-05-16:
Excellent, I will ensure this process is further clarified in games console project and any others that may need updating. Thanks for your patience.

jsteele-cnn, 2020-05-16:
I wasn't able to find this option in the RetroPie menu either, so your instructions may be based on an previous version: Once in the RetroPie GUI, choose [Manage packages] > [driver].
I'm reimaging my sd card, and then I'm going to reinstall/configure everything from scratch.
I'm not currently able to connect the controller I was sent as part of the reward to the pi at this time though.

ChristopherRush, 2020-05-16:
@jsteele-cnn Check this out here which has more detail and screenshots - https://learn.pi-supply.com/make/how-to-connect-your-bluetooth-games-controller-to-your-raspberry-pi-using-retropie/#using-the-retropie-gui-to-configure-the-ps3-controller-bluetooth-connection

jsteele-cnn, 2020-05-16:
The correct orientation is 270, not 90

ChristopherRush, 2020-05-16:
@jsteele-cnn Will check that out. It probably depends which way you have the case mounted. IR LED and buttons should be on the right hand side of the display.

jsteele-cnn, 2020-05-16:
There's only one way to mount the case because of where the ports are.

ChristopherRush, 2020-05-16:
Will double check that tomorrow, thanks.

shawaj, 2020-05-18:
@jsteele-cnn just to add, and just so you know, the Raspberry Pi HAT spec is a bit limited in that it only allows for a single HAT to be configured automatically via the EEPROM. That is why you were seeing this error before changing the PiJuice EEPROM address...as both HATs have EEPROMs which conflict with each other.

And by the way, it usually helps if you are not rude to the very people you are asking for help from in the first request for help.

Anyway, glad you are all up and running now and thanks for flagging up this issue.

jsteele-cnn, 2020-05-18:
@shawaj 
I'm sure you can imagine that having spent money to back a project before my youngest child was born and not receiving the backer reward until she was old enough to be reading, writing, and riding a bike with no training wheels, my level of confidence in the team was incredibly low.
With all the previous incompetence and perpetual failures over the years related to this project, including the fact that the writeup _specifically for it_ was wrong in a number of ways, I think my response was, if anything, mild.
If you'd like to see what a less-than-mild response looks like, I'm happy to oblige you.

ChristopherRush, 2020-05-18:
@jsteele-cnn Did you get the controller working ok?

closed 2020-10-24 by shawaj