PiSupply/PiJuice #502: Irregular battery readings

mikelinton, 2020-05-18:
I'm running a new Pijuice (purchased a few months ago) on a RPi4 with PiJuice firmware 1.4. The system is only on battery, wakes up every hour, does a few things for about 3 minutes, then shuts down with the SYS_FUNC_HALT_POW_OFF executing 30 seconds after shutdown to remove all power to the RPi. While its awake I'm reporting the battery % with a python script. The readings bounce around a lot and although they generally decline, they follow somewhat of an oscillating pattern as they decrease (see image). This behavior is pretty consistent, no matter the wake up/sleep settings I've enabled. The below graph represents about 17 hours.

[image]

I've read about somewhat similar issues with irregular battery readings but those were fixed in FW 1.3 (I think). Any ideas of why this might be happening? 

Thank you in advance.

Mike

tvoverbeek, 2020-05-18:
Since you only are active for 3 minutes per hour, you also get measurements for 3 minutes.
I cannot find this back in your graph. Could you indicate the individual measurements with a symbol (e.g. + or *)?
Also is the time for your measurements coming from the Pi or is this done externally?
When using Pi time it depends if you are on the internet or not.
With an answer on these questions I might be able to give you some answers.

mikelinton, 2020-05-18:
Hi Ton - Thanks for the reply. Responses to your questions...

- I am only taking a single battery measurement each time it wakes up, not continuously over the 3 minutes. Roughly speaking (could be off by some 10's of seconds), the first minute is boot up, then it waits 60 seconds, then takes a battery and other sensor readings, uploads those values to an externally hosted DB over a cell connection, then has a few more delays for the last minute, and shuts down.
- For clarity, below is the table of readings and time
- Its coming from Pi time, which has an internet connection

time | battery %
-- | --
2020-5-17 19:33:40 | 92
2020-5-17 20:3:39 | 95
2020-5-17 20:33:44 | 61
2020-5-17 21:3:41 | 66
2020-5-17 21:33:44 | 79
2020-5-17 22:3:54 | 84
2020-5-17 22:33:53 | 84
2020-5-17 23:3:46 | 83
2020-5-17 23:33:43 | 82
2020-5-18 0:3:43 | 43
2020-5-18 0:33:43 | 29
2020-5-18 1:3:48 | 46
2020-5-18 1:33:48 | 56
2020-5-18 2:35:31 | 62
2020-5-18 2:37:16 | 61
2020-5-18 3:33:47 | 60
2020-5-18 4:3:45 | 6
2020-5-18 4:5:23 | 19
2020-5-18 5:3:45 | 27
2020-5-18 5:33:45 | 3
2020-5-18 6:3:48 | 6
2020-5-18 6:33:44 | 16
2020-5-18 7:3:48 | 21
2020-5-18 7:33:47 | 3
2020-5-18 8:3:43 | 13
2020-5-18 8:33:51 | 18
2020-5-18 9:3:48 | 20
2020-5-18 9:33:48 | 3
2020-5-18 10:3:51 | 11
2020-5-18 10:33:49 | 15
2020-5-18 11:3:49 | 18
2020-5-18 11:33:49 | 18
2020-5-18 12:3:46 | 18
2020-5-18 12:33:51 | 17

tvoverbeek, 2020-05-18:
TIme sequence is strange. Mostly one point per 30 minutes, but a few cases with jumps
2020-05-18 01:33 - 02:35 - 02:37 - 03:33.
Do you timestamp when they are entered in the database?
Do you use the PIJuice RTC to set the system time at boot?
Pi time will not be very reliable. At boot time will start at the time of the last shutdown until synchronised by the internet when you do not use the RTC to set the time.

I can imagine when you sample the battery while the cell modem is active you get a low reading due to the power draw by the modem.

mikelinton, 2020-05-18:
Ah sorry, I misspoke in my original post, it does wake up every 30 mins for 3 mins, not every hour, and takes a battery reading. And that is a good point about sampling when the cell modem is active, let me check on a few of these things and get back to you to see if any of it makes a difference. Thank you!

mikelinton, 2020-05-19:
I simplified my test and got better results but still some odd behavior. I adjusted...
- Removed the use of the cell modem all together (and unplugged it from USB along with all other peripherals other than PiJuice obv) and only uses wifi for internet connection
- Removed a few other tasks so it just grabs battery and a single temperature reading and uploads to the DB resulting in about a ~2 minute awake period and then off for the remainder of the 30 min cycle
- Added a created_at column to the DB that is populated by the DB system time (not RPi time)

Below is the graph and table again. As you'll see, it starts out exactly as I would expect but then has a significant dip about half way through with a gradual increase back up to what would be the expected battery readings.

Let me know if any further thoughts come to mind. It might also be helpful to know more details as to how the battery fuel gauge works and what could effect the readings so to avoid those situations in the future.

Thanks again!

[image]

battery | created_at
-- | --
88 | 2020-05-19 1:01:39
87 | 2020-05-19 1:31:49
84 | 2020-05-19 2:01:53
82 | 2020-05-19 2:31:49
81 | 2020-05-19 3:01:53
79 | 2020-05-19 3:31:53
77 | 2020-05-19 4:01:49
76 | 2020-05-19 4:31:50
73 | 2020-05-19 5:01:49
71 | 2020-05-19 5:31:49
24 | 2020-05-19 6:01:53
36 | 2020-05-19 6:31:53
45 | 2020-05-19 7:01:53
50 | 2020-05-19 7:31:57
52 | 2020-05-19 8:01:51
53 | 2020-05-19 8:31:56
51 | 2020-05-19 9:01:56
50 | 2020-05-19 9:31:52
49 | 2020-05-19 10:01:57
48 | 2020-05-19 10:31:52
47 | 2020-05-19 11:01:57
46 | 2020-05-19 11:31:57
44 | 2020-05-19 12:01:55
43 | 2020-05-19 12:31:55