PiSupply/PiJuice #503: /dev/i2c-1 Missing after restarting RPI4

Shaun-Harrison, 2020-05-22:
I've got a headless RPI4 with the pijuice attached

I've installed pijuice-base as root user which installs fine, I use su pi to switch to the pi user as I've disabled ssh for the pi user. I then enter pijuice_cli which shows the config menu 

I've then found a script on another issue here which is the following 

    import pijuice
    import os
    pj=pijuice.PiJuice(1, 0x14)
    pj.power.SetWakeUpOnCharge(0)
    os.system("halt")

I've unplugged the power source to the RPI4, then ran this script as the root user, which shuts down the RPI4. I then plug the RPI4 back in and it does power up as expected, but here is where I'm finding the /dev/i2c-1 file is now missing, and when trying to run pijuice-cli as pi as before it shows the 
'No PiJuice found of Failed to use I2C bus. Check if I2C is enabled' 

Does this happen because the script has been ran as root? 
If so, I'll figure a way to run it as a different user in my setup, but how do I get pijuice back working once this happens? Do I have to uninstall all the packages and reinstall?

Shaun-Harrison, 2020-05-22:
It seems that no matter if I run the script or not after restarting the RPI4 the /dev/i2c-1 is gone

ChristopherRush, 2020-05-22:
Try the following:
```
#!/usr/bin/python3
from pijuice import PiJuice
import os
pijuice = PiJuice(1, 0x14)

# Remove power to PiJuice MCU IO pins
pijuice.power.SetSystemPowerSwitch(0)
# Set wakeup
pijuice.power.SetWakeUpOnCharge(0.0)
# Remove 5V power to RPi after 60 seconds
pijuice.power.SetPowerOff(60)
# Shut down the RPi
os.system("sudo halt")
```

Shaun-Harrison, 2020-05-22:
Hi @ChristopherRush 

I've updated my script, does my comment however about how just restarting the RPI4 without any script cause the file to go missing change anything?

tvoverbeek, 2020-05-22:
@Shaun-Harrison When you boot the Pi without the PiJuice is /dev/i2c-1 then present?
Can you list the contents of /boot/config.txt? There shouid be a line `dtparam=i2c-arm=on`.
Also the output from `lsmod | grep i2c` could give some clues.
The following should be there:
```
i2c_bcm2835            16384  0
i2c_dev                20480  0
```
The i2c-dev module is responsible for creating the /dev/i2c-1 device.
And I am assuming you are running Raspbian or are you using a different OS?

Shaun-Harrison, 2020-05-22:
Hi @tvoverbeek 

I will test shortly booting the RPI without the PiJuice attached
I am running Raspbian - however I am booting with an SD but have set up the pi to then use a external SSD as its main storage 

In /boot/config.txt I have the following (there is more but all commented out so these sections are what has stuff enabled)

```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtoverlay=i2c-bcm2708
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

```

So I can see that I have `i2c_arm=on` where as you commented `i2c-arm=on` was this a typo or should I change mine to '-'?

The output of `lsmod | grep i2c` is the same as yours initially, however after restarting the RPI4 via `reboot` this is when the issues start and `/dev/i2c-1` is gone and the output of the previous command is 

`i2c_dev                20480  0`

i2c-dev is present in my /etc/modules file in case that information is needed

tvoverbeek, 2020-05-22:
@Shaun-Harrison No typo. `dtparam=i2c_arm=on` should be with an underscore in /boot/config.txt. Also `i2c-dev` should be in /etc/modules.
When powering up the Pi (without PiJuice) it works and then after a reboot (Pi remains powered) the i2c device is gone. Correct?

The only thing I can suggest is to look at the dmesg output to see if there are any error messages regarding i2c. Any other i2c devices connected?

Shaun-Harrison, 2020-07-27:
Hello, I had to reinstall my SSD over the weekend and this issue is no longer a problem

closed 2020-07-27 by Shaun-Harrison

hferentschik, 2022-11-15:
I just had the same issue. Drove me insane debugging it. I have a project with Pi 3 + PiJuice + LoRa HAT. Worked fine for several months. I tried to make some changes, and then suddenly, nothing worked, and the i2c devices disappeared. Kept backtracking, but even the original code would not work anymore. In the end, I flushed a new image into the SD card and it started working again. I guess something during my experiments corrupted something in the boot config. Unclear really.