PiSupply/PiJuice #504: Normal operating conditions

lg007, 2020-05-24:
Hello,

I am trying to find the normal operating conditions of the PiJucie + onboard battery combination. I would like to know what are the minimum and maximum temperature and humidity conditions that the board and its supplied battery is supposed to work safely.
I assume there must be a document containing this information. I suppose this must be published to be able to declare CE conformity and to conduct the safety tests for the board and the supplied battery (which are available anyway in the github repo). 

Can anyone help me on this?

Thanks,

tvoverbeek, 2020-05-24:
Note there is compliance information in https://github.com/PiSupply/PiJuice/tree/master/Documentation/Compliance
(mainly radiated emission)
For the batteries:
https://github.com/PiSupply/PiJuice/tree/master/Documentation/Battery%20Safety

For operating conditions just look at the battery profile.
For the BP7X:
Cold temp  0
Cool temp 2
Warm temp 50
Hot temp 70
All temperatures in dec C.

lg007, 2020-05-24:
Thanks for the fast answer.

Does this mean that I can have the PiJuice + Onboard Battery lets say for 30 minutes working at an ambient temperature of 58°C (working on 5V DC, not from the battery)?

tvoverbeek, 2020-05-24:
Yes, that should be OK.
Note that the PiJuice sits on top of the RPi. Especially with the Pi4 this causes already a
higher temperature for the battery than the plane ambient temperature.
When outside the cold-hot temp range no charging occurs.
Between cold and cool (and warm and hot) the charging current is reduced.

lg007, 2020-05-24:
Thanks again for the fast response. 

I can see that you have planned this circuit for everything. Congratulations again!

closed 2020-05-24 by lg007