PiSupply/PiJuice #505: Repeated wakeup after boot on charge

mikelinton, 2020-05-27:
My setup is turning on every 30 mins (30 min period setting), doing a few things for 2-3 minutes and then shutting off. Upon boot, I'm setting the wakeup flag with a cron job as well as setting the system time from the RTC. All of this is working successfully. I also have it set to shutoff when charge reaches 20% and boot when charge gets to 80%. Both of these also work successfully but after it boots after reaching 80%, it does not resume its 30 minute boot cadence and I do not get a repeated wakeup again. I have to use the switch to boot it again and the 30 min cadence resumes as expected. And from what I can tell from the syslog the wakeup flag is being set on every boot.

Any ideas why this might be happening or some other setting I'm missing?

Mike

tvoverbeek, 2020-06-07:
@mikelinton have been lookin at this and trying to reproduce it. 
One question: When you shutdown for low charge howdo you make sure the 30 minute wakeup cadence does not continue? The RTC keeps running while the RPi is shut down.
Do you clear the wakeup flag when shutting down for low charge?

shawaj, 2020-10-24:
@mikelinton any update here?

mikelinton, 2020-10-26:
I don't believe I was clearing the wakeup flag when shutting down for a low charge. I'd be curious why clearing the wakeup flag would then allow it to startup when it reaches above 80% charge?

I've actually successfully had my system running out in the woods for the last 4 months successfully waking up every hour (adjusted from every 30 mins) but the charge has never dropped below 80% so I've avoided this issue for now. I will hopefully be recovering the system in a few weeks to avoid the winter and will be able to spend some time with it then.

shawaj, 2020-10-26:
@mikelinton have you got it charging with solar to keep the battery charged or does it only turn on briefly? 4 months is pretty good going. Is that with the 12000mAh battery?

mikelinton, 2020-10-26:
Yes, I have it on the smaller/standard battery on the pijuice but then also a much larger battery (~20,000 mAh) that is charged with solar that is then charging the pijuice battery. Wakes up every hour and is on for about 2-3 mins max then shuts down. Have not seen the pijuice battery dip below 99%. It's working surprisingly well and reliable.

shawaj, 2020-11-03:
@mikelinton so just to check - you don't clear the wakeup flag before going down for a low charge shutdown?

Do you have your script by any chance?

Good to hear that this hasn't happened in practice, but good to solve it in any case :-)