PiSupply/PiJuice #506: COMMUNICATION_ERROR in GUI - No addresses on I2C Bus

matthewchicky, 2020-06-07:
Hello,

To begin, I will state that I've viewed the other related Issues but none of the fixes shown there have resolved my issue.

I'm attempting to access my new PiJuice HAT in the 'PiJuice Settings' gui on my RPi 4B 4GB. However, I am getting COMMUNICATION_ERROR messages throughout in the gui. Further, the battery icon indicator is not present, which I believe should be visible in the top-right corner of my screen, on the task bar.

The hardware of the PiJuice HAT appears to be working fine--the BP7X has charged successfully, will power the RPi when I remove the USB-C charger, and the SW1 switch will power up the RPi both with and without the USB-C charger connected. The DIP switches are set for BP7X.

I have added dtoverlay=i2c-bcm2708 to /boot/config.txt, as instructed at install of pijuice-gui and in the other resolved issues. Also, command lsmod shows i2c-bcm2708 in the module list, with '0' listed under the 'Used By' column.

I confirmed using sudo raspi-config that ARM I2C is enabled. **Running i2cdetect -y 1 shows no addresses on the I2C bus with the PiJuice connected** (there are dashes throughout the address space). 

Another note: the title of my PiJuice gui is 'PiJuice Settings'. In #268 it was stated that the gui title may need to be 'PiJuice Settings (Python 3 version)'. Not sure if that's problematic.

Raspbian OS and PiJuice GUI versions:
```
pi@rocket:~ $ apt list --installed 'pijuice-*'
Listing... Done
pijuice-base/testing,now 1.6 all [installed,automatic]
pijuice-gui/testing,now 1.6 all [installed]
pi@rocket:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

Hoping that someone can help me with this so we can get the PiJuice talking to the RPi. Thanks for your time.

tvoverbeek, 2020-06-07:
@matthewchicky Obviously there is no i2c communication at all since `i2cdetect -y 1` shows nothing.
The recent batch of PiJuice has firmware version 1.4 already installed, so the work-around with `dtoverlay=i2c-bcm2708` is not needed.
The current package version (1.6) is python 3 only, therefore the '(Python 3 version)'  in the window title has been removed.

First we need to get basic i2c communication going (i.e. i2cdetect should detect something).
Does device /dev/i2c-1 exist?
What is the output from `lsmod|grep i2c`?
My output (on a Pi4B-2GB)is:
```
pi@raspberrypi:~ $ lsmod|grep i2c
i2c_bcm2835            16384  3
i2c_dev                20480  6
pi@raspberrypi:~ $ 
```
The i2c-dev module is responsible for the creation of the i2c devices.
Without it i2cdetect and pijuice will not be able to communicate via i2c.
An other thing to check is if the PiJuice is properly connected to the 40 pin GPIO connector
(not shifted by 1 row or column, have made this mistake this myself).

matthewchicky, 2020-06-07:
@tvoverbeek Thank you for your prompt response.

I have commented out `dtoverlay=i2c-bcm2708` in the config file, since it's not needed. Then I rebooted.

Here is the output you requested:
```
pi@rocket:~ $ lsmod|grep i2c
i2c_bcm2835            16384  2
i2c_dev                20480  4
```
```
pi@rocket:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --       
```
Another command that may be helpful:
```
pi@rocket:~ $ ls -la /dev/i2c*
crw-rw---- 1 root i2c 89, 1 Jun  7 12:03 /dev/i2c-1
```

tvoverbeek, 2020-06-07:
The lsmod output looks fine.
The only difference with mine is that i2c-bcm2835 is referenced 2 times by you and 3 by me
and i2c-dev is referenced 4 times by you and 6 by me.
RPi4 has 6 i2c interfaces. Have you reassigned any of them using overlays?
Also if you run i2cdetect as root (`sudo i2cdetect -y 1`) do you get any output?

matthewchicky, 2020-06-07:
`sudo i2cdetect -y 1` does not show any output, same as running the command without sudo.

I haven't reassigned any of the interfaces that I know of...I'm relatively new to RPi and aside from enabling the camera, i2c interfaces and installing Spyder3 I haven't really done anything with the pi.

Do you think that the fact that I'm only referencing 4 i2c-dev interfaces instead of 6 this could be causing the issue? Could a fresh OS install remedy this?

I'm tempted to reinstall the OS on the microSD afresh and seeing if that resolves the issue...

matthewchicky, 2020-06-08:
@tvoverbeek The fresh install of Raspbian did the trick! I installed on a new, 8GB microSD and after the initial setup of the OS, ran `sudo apt-get install pijuice-gui` to get the software as the instructions state. Then everything worked correctly.

Wish I knew why it wasn't working with my 32GB microSD....but I'm glad there isn't a hardware issue with the HAT. Something odd: when I run `i2cdetect -y 1` with this new installation, the address field populates almost instantly in the terminal window. On the old installation, the same command would step through each possible address in the i2c address field at a rate of about 1 per second. So something seems to be different with the i2c bus.

`lsmod|grep i2c` still gives the same output as before, however. But ultimately, I'm just glad it's working. Thanks for your help.

closed 2020-06-08 by matthewchicky

tvoverbeek, 2020-06-08:
Glad it is working now. Maybe a problem with your 32GB SD card.
Suggest to thoroughly erase it (e.g. with SD card formatter, https://www.sdcard.org/downloads/formatter/) before using it again.