PiSupply/PiJuice #507: What to change in configuration is battery is removed

lg007, 2020-06-15:
Hello,

This is really a question, not an issue.

I have a setup with internal battery that has been running for a month without any problem. The installation will be changed in a way that the PiJuice will be connected to an UPS, so the internal battery will not be needed anymore. For the sake of easier maintenance, I am planning to remove the internal battery. 

I am currently using “watchdog”, “wake up on charge” and “minimum battery voltage” system tasks. From these only “watchdog” has any meaning if the battery is removed (I guess the “watchdog” will still be functional as I am powering the RPi through the PiJuice).

Will I have to change anything in the settings of the PiJuice, or I can leave it as it is? 

Regards,

tvoverbeek, 2020-06-15:
@lg007 I would disable the "min battery voltage" ant the "Wake up on charge" on the system task
(Watchdog the only item enabled + system task). On the HAT configuration enable the "No battery turn on".

lg007, 2020-06-15:
Thanks a lot for the fast and accurate answer!

lg007, 2020-06-22:
Hello @tvoverbeek,

I have done the settings as you have suggested. However after removing the battery, the system started to reboot on random intervals. Since I don't really know what is ongoing, I have some questions.
- Is the Pijuice monitoring the load and the voltage going to the RPi (even rebooting it when this goes above certain limit)? Could the reboots happen because of this? (the power supply feeding the Pijuice is 5V 4A, so there should be enough power going into it).
- As a workaround until I find the problem, what happens if I shut down the pijuice service? Will the system be functional (access to the power on button of the Pijuice is very difficult). I mean will it switch on automativally?

Before removing the battery everything was working fine.

Thanks,

tvoverbeek, 2020-06-22:
If you power the Pi via the PiJuice (external power connected to the PiJuice) then the power supply capacity does not play a large role, since there are two power converters on the PiJuice (From external to internal power 3.3V, and them from 3.3 to 5V  on the GPIO).
As far as I remember the PiJuice can only supply max 2.5A on the 5V GPIO pins.
This is a bit meager for a Pi4 especially if it has connected peripherals like an LCD display.

Yes, there is a emergency shutdown when the power drawn from the PiJuice is too high.
All the normally programmed shutdowns (SYS_FUNC_... and via user functions) require the service. So disabling the service will eliminate these shutdowns.

Cleanest solution is to connect your UPS 5V output directly to the Pi, remove the PiJuice battery, and disable the pijuice service.

lg007, 2020-06-22:
Thanks, I still have a couple of questions:

> Yes, there is a emergency shutdown when the power drawn from the PiJuice is too high.

Is this in the firmware of the MCU or in the python script?

> Cleanest solution is to connect your UPS 5V output directly to the Pi, remove the PiJuice battery, and disable the pijuice service.

I still don't understand why everything was working fine when the battery was still in place. Furthermore, by disabling the service I will loose the watchdog functionality as well (I guess the watchdog only works if the service is running plus the RPi is powered through the PiJuice).

And my last question: Suppose I leave the 5V connected to the PIjuice to power the RPi through GPIO and I disable the pijuice service, will the system power on automatically if the 5V is taken away and switched back? Or to phrase it differently, is the "No battery turn on" functionality available if the pijuice service is not running? (but was set ON before disabling the service)