PiSupply/PiJuice #508: Script not running on PowerLoss

geoffr98, 2020-06-16:
I am trying to run a script when power is lost, but it isn't running. I suspect I have something wrong I'm hoping someone can point out. The script doesn't seem to be running as my results file remains empty.  If I run the script without the PiJuice or halt command it works fine from a ssh terminal..

PiJuice Settings:
- SystemEvents -> No Power -> User_Func1
- User Scripts ->  User Func1 -> /home/pi/PiJuice_Shutdown.py

Script 
-rwxr-xr-x  1 pi   pi    802 Jun 15 22:09 PiJuice_Shutdown.py

`#!/usr/bin/python3

import os
import time
from pijuice import PiJuice

os.system("echo 'PiJuice Shutdown output \n'  > /home/pi/result")

pj = PiJuice(1, 0x14) # Instantiate PiJuice interface object

# Shutdown OpenCPN
ret1 = os.system("killall opencpn")
msg = "killall opencpn \n" + str(ret1) + "\n"
cmd = "echo '" + msg + "' > /home/pi/result"
os.system("rm result")
os.system(cmd)
while(os.system("pgrep -x opencpn") == 256):
        time.sleep(1)
        os.system("echo 'wait \n'  > /home/pi/result")

time.sleep(10)
print("Shutting down")
os.system("echo 'Shutting Down \n'  > /home/pi/result")

#Remove Pi power after 30 seconds
pj.power.SetPowerOff(30)
os.system("echo 'pj(30) n\n'  > /home/pi/result")

# Shutdown
os.system("sudo halt")`

tvoverbeek, 2020-06-16:
@geoffr98 Can you attach your script as a file?
The github markup has mangled your script.

Anyway in its current form I see a back-tic (`) just before the #!/usr/bin/python3.
If it is really there, the script will not run because the first line is not recognized.
Does the script run manually when you start it as "/home/pi/PiJuice_Shutdown.py"
(no python3 in front)? It should not if the back-tic is really there.

geoffr98, 2020-06-16:
I'm not connected to the Pi right now, but I've attached what the post should have had for the code. The python file runs fine when I run from ssh  (eg  ./PiJuice_Shutdown.py ).

[PiJuice_Shutdown.zip](https://github.com/PiSupply/PiJuice/files/4787102/PiJuice_Shutdown.zip)

ChristopherRush, 2020-06-16:
Who is the owner of the script? I find that when creating files via SSH the
owner is root and not the current user. You can check with command `ls -l`.

geoffr98, 2020-06-16:
Owner of the script is pi:
-rwxr-xr-x 1 pi pi 802 Jun 15 22:09 PiJuice_Shutdown.py

tvoverbeek, 2020-06-16:
@geoffr98 Your script (from the zip file) looks fine and is executable.
Have you enabled the "System Task'? If not, your script will not be executed when there is no power input.
Also the pi user has to belong to the pijuice group. Normally this is taken care off by the pijuice-base installation.

geoffr98, 2020-06-16:
@tvoverbeek 
Yes the System Task -> system task enable  checkbox is checked.
Pi is in the pijuice group

tvoverbeek, 2020-06-16:
OK. Last thing to check: is the pijuice service enabled?
What is the output from `systemctl status pijuice.service`?
O, and which OS version?

geoffr98, 2020-06-16:
[image]

[image]

geoffr98, 2020-06-16:
Might be the OS... I completely forgot.. I'm on the new 64 bit OS - although it should still work I would think?

tvoverbeek, 2020-06-16:
On 64-bit the pijuice_cli and pijuice_gui wrappers are 32-bit executables. (They force the python3 interpreter to run as user pijuice).
You can work-around it by running the cli/gui manually:
`sudo -u pijuice python3 /usr/bin/pijuice_cli.py`. Similar for pijuice_gui.

geoffr98, 2020-06-16:
I see the same information if I run the command you show above and if I use the installed GUI from the menu.  (so probably not why my script isn't running correctly? )

geoffr98, 2020-06-16:
My test is as follows:
1. Apply external power (Causes Pi to boot up)
2. Remove Power (script should run)
3. Wait a couple minutes
4. Apply external power again
5. Check the result file (should be populated... but so far isn't)

geoffr98, 2020-06-16:
I just commented out the last lines of the script (SetPowerOff(30) and sudo halt). 
When I remove the power the Pi Shuts down - shouldn't it stay on if a user script is running and I've commented out these lines?

tvoverbeek, 2020-06-17:
Are you sure your script is running with the lines removed?
Do you get anything in home/pi/result? (By the way I would use >> instead of > in the echo statements)
If the script is not doing the shutdown, something else does.

geoffr98, 2020-06-19:
Is there a way to confirm that the PiJuice is controlling the shutdown?  (Wondering if when I moved the Juice from the Pi3 to Pi4 I missed something...)

tvoverbeek, 2020-06-19:
@geoffr98 Came to think of something.
SInce you are running OpenCPN (maritime charting software) does the PiJuice have to supply both the Pi4 and e.g. the 7 inch display?
In that case the capacity of the supplied BP7X is not sufficient for supporting the combination for a long time. It will probably do an emergency shutdown because it draws too much power, even if the battery is still at e.g. 70%.

If the Pi is shutdown by one of the SYS_FUNC_* functions LED D2 will be flashing red/green just before the shutdown command.

shawaj, 2020-10-24:
@geoffr98 any update on this?

geoffr98, 2020-10-27:
Aaron,

I still have this problem - unfortunately I haven't had time to continue to
dig into it. Hopefully this winter work will calm down a bit so I can
figure it out.

Geoff.

shawaj, 2020-10-27:
@geoffr98 ok let us know when you've had some time to try our suggestions

geoffr98, 2020-12-26:
@tvoverbeek 
I've finally been able to get back to this (Brought the Pi home from the boat) To answer your questions above:
- The screen is not powered from the Pi
- When I remove the Power to the PI I do not see the LED 2 come on (or anything)
- LED1 is set as Charge Status and Led2 is set as User_LED

tvoverbeek, 2020-12-26:
@geoffr98 First the LED2 thing is my bad. It only flashes when forcing a parameter reset.
Your LED1 and LED2 settings are the default ones.
You write: The screen is nit powered from the Pi.
So I assume it is powered separately. It could be that is is back-powering the Pi/PiJuice via the GPIO 5V pins.
If so when the external power to the Pi goes off the 'no power' event will not be detected.

I am still wondering why yiu can start the gui from the menu.
Did you rebuild the pijuice_gui and pijuice_cli wrappers fro 64-bit and installed them?
Or are you running a 64-bit kernel with 32-bit userland?

geoffr98, 2020-12-26:
@tvoverbeek 
Here at home the screen does not loose power, just the Pi .. and the Pi powers off, so there doesn't appear to be any back-powering going on.
I did not rebuild the PiJuice_GUI/CLI - I'm pretty sure I just did the apt-get.
I would assume my user land is 64 bit as I haven't done anything other than use the standard image for the Rpi4

tvoverbeek, 2020-12-26:
@geoffr98 So we still do not know why the Pi powers off since the 'no power' script does  not run (or does it run now at home?)

geoffr98, 2020-12-26:
@tvoverbeek 
Yes, still don't know why the script isn't running - I think PiJuice isn't getting the shutdown or isn't powering the pi..
I have the following:
- Powered through the GPIO pins 
- USB cable between PiJuice and Rpi

[image]

shawaj, 2020-12-26:
Why do you have a USB cable connecting the PiJuice to the Raspberry Pi USB?

The micro USB on the PiJuice is a power input.

shawaj, 2020-12-26:
Also @geoffr98 further to my last message, you say you are powering it through the GPIO pins - what power source are you using for this?

geoffr98, 2020-12-27:
@shawaj I don't remember exactly why I have the USB cable between the PiJuice and Rpi - I needed it when I had this working on my Rpi3 if I remember correctly.. but it doesn't seem to make a difference now.

Yes I am powering the Rpi (and PiJuice) through the 5V pins on the GPIO. I'm providing 5V (up to 3A) through a UBEC from a 12V battery (normally, right now a wall wart).

shawaj, 2020-12-27:
@geoffr98 ok. Just so I'm clear, what are the remaining issues?

I just want to make sure I'm not getting confused.

geoffr98, 2020-12-27:
@shawaj The issue is that my shutdown script is not being called/run by PiJuice when power is lost... From what I can tell the PiJuice isn't doing anything as when power is lost it shuts down immediately (as a normal Pi would when power is removed).

shawaj, 2020-12-27:
It sounds like maybe the battery isn't connected properly. Is it definitely making contact with the battery connector?

tvoverbeek, 2020-12-27:
@geoffr98 Your power connections are strange.
You power via the GPIO pins, so it supplies power to the PiJuice and RPi simultaneously.
Your extra USB cable (probably) supplies 5V from the RPi USB output to the PiJuice microUSB power input.
When you remove the GPIO power I do not know what happens (is the PiJuice battery then supplying power, or is there
a power interruption to the Pi ????)

I would remove the extra USB cable and supply your UBEC power to J4 on the PiJuice and let the PiJuice supply power
to the RPi via the GPIO.
My guess is that the 'no power' event will then be detected properly.
I do not know how good your UBEC is. When supplying power via the GPIO the 5V needs to be clean. If there is a large ripple it will cause problems. Normally the power management chip on the Pi takes care of 'cleaning' the 5V, but the GPIO power is connected after the power management chip.

geoffr98, 2020-12-27:
@shawaj - The battery is connected and shows a percentage charge on the PiJuice Icon in the task bar
@tvoverbeek - I have tried the following:
 - just remove the USB cable between the Pi and Juice (still applying 5V to GPIO) - no change in behaviour
 - No USB cable, apply power to J4 of PiJuice - Pi does not power up.
 - No USB cable, apply power to J4 of PiJuice (Pi not on), apply power to Pi USB (powers up), remove USB power (Pi still running), Remove J4 Power (shuts down, but shutdown script is not run)
 
The UBEC I'm using is this one: https://www.adafruit.com/product/1385

geoffr98, 2020-12-27:
@tvoverbeek Wake on Charge is enabled and set to 5% (Battery is showing 65%)

shawaj, 2020-12-28:
You only need to provide power to one or the other of the PiJuice or the Pi itself. Not to both.

If the battery is showing a percentage - does it also show that it's charging?

Can you start the Pi from battery with no power supply attached by pressing the button SW1?

geoffr98, 2020-12-28:
@shawaj No SW1 does not power up the Pi (with battery connected and no other power source)

When I supply power to J4 on the PiJuice the GUI says "Charging_from_in"

shawaj, 2020-12-28:
That sounds very strange.

Have you tried the factory reset procedure or reloading the firmware?

geoffr98, 2020-12-28:
@shawaj I was just wondering about that tonight - I'll try it and let you know..

geoffr98, 2020-12-28:
@shawaj 
I've reset the firmware (long press of SW1 & SW2, now when applying power to J4 the Rpi gets into a continuous boot loop. If I apply power to the Rpi it will then boot. I still can't boot by pressing the switch with just the battery. (LED glows red for a bit and the two LEDs below on the pi briefly flash)

Once it is powered up I can shut down with a long press of SW1

As an aside the same UBEC powers the Pi just fine through the GPIO. (Just not J4). Other than the Pi the only things connected are a wireless mouse and the HDMI monitor.  What is odd is that sometimes when I boot the battery shows above 60% and other times it's down at 6%.. If I remove the battery and put it back in when it says 6% it then shows above 60% again..

As a reminder (in case it triggers any thoughts) this is a Pi4 64bit OS

geoffr98, 2020-12-29:
To test that the PiJuice was still working I moved back to my Pi3b which was working before. (with an older version of the firmware and the Pogo pin, but now with the newer firmware and no Pogo pin)

- Powering to the PiGPIO pins only --> script does not run (in fact board shuts down immediately)
- Powering to the PiJuice JP4 ins --> Board is not stable, hangs and sometimes reboots 
- Powering to the PiGPIO pin with a USB cable between the Pi and PiJuice --> script does not and and shuts down immediately
- Power to the PiJuice USB ---> boots fine and seems to shutdown fine, but script doesn't run when power is removed, although there is a good delay before it shuts down.

Since I wasn't getting the script running I wrote a very simple script for SW3, pressing SW 3 did not run the script, so scripting works.

From the tests above it looks to me like the PiJuice is not seeing the power going away or is not correctly calling the script...

geoffr98, 2021-02-20:
I finally got back to this and have it working (on a Pi3, will move to Pi4 later).
I ended up flashing the older V1.3 firmware and setting things up as follows:

- Power to the Pi through J4 on the PiJuice 
-  Enable System  Task
- Wake on charge 5%
- Minimum charge 10% (when this and battery voltage were off  it wasn't working)
- Minimum batter voltage 3.3
- System Event - no power - USER_FUNC1  (my script)
- System Event - Button power off - USER_FUNC1  (my script)

PiJuice is set to detect the power source

closed 2021-02-22 by geoffr98