PiSupply/PiJuice #509: Loss Power Shutdown Delay

paulmona, 2020-06-18:
How can I delay shutdown by X seconds when power is lost on Micro USB?  Currently, the system shuts down immediately.  I currently have system event "No Power" set to SYS_FUNC_SYS_OFF_HALT but I can't see anywhere to define a delay for that.  I must be missing it as that's pretty basic functionality for UPS.  :)

tvoverbeek, 2020-06-19:
Use SYS_FUNC_HALT_POW_OFF instead. This keeps the 5V to the Pi on for 60 seconds before removing it.
If you want a different delay, use a user function to do the shutdown and enable 'Software Halt Power Off' on the system task tab and set the delay you want (minimum 20 seconds).
This delay also works if you shut down from the GUI or the command line.
SYS_FUNC_SYS_OFF_HALT does not have a delay (as you found out).

paulmona, 2020-06-19:
Thanks for the quick response. With user function and software_halt_pwt_off would I not have to monitor the charge state to stop the shutdown from happening on power restore? 

I want the Pi to shutdown after a defined delay if power supply is removed, if power is restored within the delay window then shutdown should be cancelled.

From what I can tell to get this functionality my user script would require the use of the GetStatus() to get power input status and cancel the shutdown. 

It's that correct? Kind of surprising given the behaviour in describing is pretty much UPS 101.

That said the flexibility of PiJuice is amazing. Runs my 5" touch display Pi for about 1 hour on battery. Love it!

tvoverbeek, 2020-06-19:
> I want the Pi to shutdown after a defined delay if power supply is removed, if power is restored within the delay window then shutdown should be cancelled.
From what I can tell to get this functionality my user script would require the use of the GetStatus() to get power input status and cancel the shutdown.

Yes, if you want a delay before the shutdown is started you have to implement that in a user function which is triggered by the 'no power' event. Do not start the shutdown from your user
function when you detect power has returned (by polling with the GetStatus() function) during the delay.

The delays in my first reply are started when the shutdown has been initiated.

paulmona, 2020-06-19:
Cool thanks. I'll whip up a script. 

Be a nice feature request to not need a custom use script to implement a shutdown delay on power loss.

closed 2020-06-19 by paulmona