PiSupply/PiJuice #510: trouble with 1.4 Firmware update

mattgrob, 2020-06-20:
I tried to update the firmware to 1.4, following the CLI instructions.

The command I issued was this:

pi@raspberrypi:~ $ pijuiceboot 14 PiJuice-V1.4_2020_01_17.elf.binary

Initially it seems to work - several info screens rolled by and it said completed (i didn't save the screens unfortunately).

But after this the pijuice now seems to be bricked: pressing s1 causes nothing, no leds show, no activity.   

How can I recover from this?  willing to revert back to 1.3 if helpful.

thanks.

Matt

mattgrob, 2020-06-20:
I was able to recover - I found helpful instructions here: https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md

I reverted back to 1.3 and that worked. So then I tried once again to upgrade to 1.4 and this time it worked.  I'm not sure what happened the 1st time, but one thing I hadn't done was stop the pijuice service.

Here are the steps I followed:

https://github.com/PiSupply/PiJuice/blob/master/Firmware/README.md

PiJuice bricked after successful firmware update

To recover from this issue follow these steps:

Make sure that you Raspberry Pi is shutdown

Disconnect all power to the PiJuice or Raspberry Pi and remove the PiJuice battery

Hold down SW3 button on the PiJuice and connect power to the Raspberry Pi

Release SW3 button after few seconds to enter bootloader mode
NOTE: LEDs on the PiJuice will remain off

Stop the pijuice service with: sudo systemctl stop pijuice.service

Download the Github repository sudo git clone https://github.com/PiSupply/PiJuice.git

Enter the Firmware directory cd PiJuice/Firmware/

Run pijuiceboot 14 PiJuice-V1.2_2018_05_02.elf.binary

Wait until the Firmware has successfully flashed

Shutdown the Raspberry Pi and remove power

Connect the battery to the PiJuice and boot up as normal

tvoverbeek, 2020-06-21:
@mattgrob I guess you know the normal way to upgrade is using the CLI/GUI.
Anyway, glad you got it sorted out.
Closing the issue.

closed 2020-06-21 by tvoverbeek

mattgrob, 2020-06-21:
thanks for your reply.  I did originally try the firmware update using pijuice_cli first, but it said no update needed.   That's why i then tried the pjuiceboot method.       I believe pijuice_cli said no update needed because it too was out of date.    So then I did the command line method which eventually worked, then I learned pijuice_cli could be updated - so I did the apt-get etc. to update that.

How can you tell the version of pijuice_cli that you are using?   I tried things like pijuice_cli -version and such but no luck, also couldn't find a version number on the various screens (maybe i didn't look in the right place).

Anyway, all is well know, thanks again for your reply.

tvoverbeek, 2020-06-22:
@mattgrob Look at the version of the pijuice-base package:
e.g. `dpkg -s pijuice-base`.
Current version s 1.6.