PiSupply/PiJuice #511: Additional, non-Pi board compatibility

tristinDLC, 2020-06-22:
Your documentation mentions that the PiJuice is possibly compatible with additional controllers outside of the Pi family. I'm attempting to build a custom PCB that utilizes multiple switches, either an OLED or eInk display, and hopefully a PiJuice setup all utilizing an nRF52840 run microcontroller that has the same pinout as a [3.3V Pro Micro](https://www.sparkfun.com/products/12587). Among other features, it has built-in Bluetooth and pin-outs for 100mAh LiPo charging so it's meant to be used as a wireless setup.

I get that I might not be able to utilize all of the advanced features of the PiJuice due to not having the Pi's interface, but I'd be content with the basic charging protection and just overall battery type (instead of using AAAs or a different multi-cell LiPo).

Any luck getting the basics of the PiJuice to adequately run and be recharged off a microcontroller similar to the Pro Micro?

tvoverbeek, 2020-06-22:
Something like this: https://www.adafruit.com/product/4062 ?
Will be quite complicated. PiJuice delivers 5V and this is a 3.3V device.
PiJuice can be controlled via I2C, but you need to rewrite all the low level code.
You are better off with using the on-board microcontroller functionality and a larger LiPo battery.

shawaj, 2020-10-24:
Closing this issue now as has been dormant for some time. Please feel free to reply if you do not agree

closed 2020-10-24 by shawaj