PiSupply/PiJuice #512: Recover from Power Failure

carterhickman, 2020-06-24:
Hey Pi Supply! 
My name is Carter Hickman, I'm an electrical engineer for a company called Roto, based out of Dublin, OH. We make interactive exhibits for science centers and museums. 

I just received a Pi Juice for one of my exhibits. I've never used one before. I need the Pi Juice to shutdown my Pi4 every night at 6 PM, and start up the Pi every morning at 6 AM. In between those times, when the museum is busy, I need the Pi to be reliant on the Pi Juice in case of any power outages. My Pi WILL NOT be in read-only mode therefore data corruption on the SD card could kill the OS image. 

Is this possible to do on the Pi Juice? If so, could you give me some pointers as to how to do so? Can I power down without using the "crontab" solution? I've been having trouble figuring out how to do this on my own using the online resources I have found and would greatly appreciate some help!

I don't think source code is helpful in this case as I would just like some general tips but if you need anything, let me know! 

Thanks!

tvoverbeek, 2020-06-24:
@carterhickman Hee some thoughts about how it could be done.
I assume you want to use the PiJuice to do an orderly shutdown in case of power loss.
Since the Pi4 is more power hungry than previous Pis and since also some power using peripherals could be attached I
assume the external power is connected to the Pi4 USB-C connector.

To get the start at 6am use a recurring alarm at 6am every day. This alarm is provided by the (simulated) RTC on the PiJuice.
However if your Pi4 is connected to the internet there is a complication: The timesyncd daemon (which synchronizes the time
from the internet with the system clock) also updates the RTC **and disables the alarm**.
To avoid this change the ID-EEPROM address to 0x52. Then the OS will not read the device-tree fragment from the ID-EEPROM and hence not interfere with the PiJuice alarm function.
To still get the time synchronization functionality add the line `dtoverlay=i2c-rtc,ds1307` to /boot/config.txt.
The ds1307 has the same time fnctions as the (simulated) ds1339 **but no alarm function**.

For the shutdown at 6pm you will still have to use a cron-job

For the power loss, set the 'No Power' event to execute SYS_FUNC_HALT_POW_OFF
Enable Wakeup-on-charge with a low level (0 or 1%). This will ensure the Pi reboots when external power returns.

There probably are some corner cases which need to be handled, but this is the basic idea.
For more nitty gritty read the Software Readme: https://github.com/PiSupply/PiJuice/tree/master/Software

carterhickman, 2020-06-30:
Thanks so much! Automatic start up and timed shut down via cron tab are working. I followed your steps for a power loss event, and tried it out by pulling power. My only problem is that I'm not sure how to verify that a safe shut down is taking place. When I pull power, my monitor running the GUI goes black immediately. Is that what is supposed to happen? How can I verify that safe shutdown is taking place? Thanks again. 

P.S. I don't have to attach the pogo pin, correct?

tvoverbeek, 2020-06-30:
@carterhickman Yes, you do not need the pogo pin.
Assuming you have connected SYS_FUNC_HALT_POW_OFF to the 'no power' event then you can watch the D2 LED
on the PiJuice and the Leds on the Pi.
AT shutdown the D2 Led will flash red/green and the green led on the Pi will show activity and end with 10 short flashes.
After 60 seconds the red power LED on the Pi should go off and the PiJuice to a low power state (slowly blinking weak led)

carterhickman, 2020-06-30:
Gotcha. That is happening, so I can verify this is working! Thanks so much for the help.

closed 2020-10-24 by shawaj