PiSupply/PiJuice #514: Makerkit Point-Shoot-Cam is Python 2 only

tvoverbeek, 2020-07-01:
The Point-Shoot-Cam is Python2 only and thus cannot use the PiJuice module because the PiJuice code is Python3.
Hence there is no battery indicator.
The C-module used to convert yuv to rgb (from Adafruit) is also Python2 only.

I am working on porting the yuv2rgb C module for Python3 compatibility.
Next is to check the Python code for Python3.

rockman55, 2020-07-02:
One of the issues I am having is when I try to modify the size of the picture taken using a HQ Pi Camera on the following piece of code :
sizeData = [  # Camera parameters for different size settings
    # Full res      Viewfinder  Crop window
    [(4056, 3040), (320, 240), (0.0, 0.0, 1.0, 1.0)],  # Large
    [(1920, 1080), (320, 180), (0.1296, 0.2222, 0.7408, 0.5556)],  # Med
    [(1440, 1080), (320, 240), (0.2222, 0.2222, 0.5556, 0.5556)]]  # Small
Notice that the Large size has been changed to 4056, 3040, when this size is selected in the viewfinder screen and a picture is taken then the program goes into a loop where all you see is the Working screen over and over.
If there is anything I could do to help please let me know and thanks for opening this.

tvoverbeek, 2020-07-03:
@rockman55 For taking a picture the python picamera module is used.
See the takePicture function lines  527-529. I suspect the camera.capture command throws an exception
which then results in a continuous busy loop.
Try taking a picture using a test python script with  only the picamera module and see if you get an error using the same parameters for resolution and crop.

 I do not have a HQ camera, but in the RaspberryPi forums there are reports the HQ camera works with picamera.

tvoverbeek, 2020-07-03:
@rockman55 Could you run this small pythone script:
```
from picamera import PiCamera

camera=PiCamera()
print(camera.revision)
```
For the V1 camera it returns `'ov5647'`.
For the V2 camera `'imx219`.
For the HQ camera I expect `'imx477'`.

Also look at the available modes for the 3 camera types in https://www.raspberrypi.org/documentation/raspbian/applications/camera.md
But maybe you already did.

rockman55, 2020-07-04:
I have been able to correct the problem with the step of taking a picture  with the resolution set at 4053,340 for the new Raspberry Pi HQ cam. It turned out to be a simple fix, by just increasing the Pi's memory for 128 to 256. I have not tested any other memory size but you might want to test a lower number if you do not have enough memory on your pi. The Picamera module documentation states that there is a minimum memory size for the Pi Camera V2.1 of 128, but I could not find any requirement for the HQ camera. Good Luck. Thanks to tvoverbeek for helping me resolve this problem.

tvoverbeek, 2020-07-04:
@rockman55 Good you got it working.
I assume you mean video memory at 256MB.
I am still interested in the output of camera.revision (see above).

rockman55, 2020-07-04:
I had sent this as an email please let me know if this is what you wanted to see from the camera.revision:
I ran your script with the 256 memory and I have the new HQ camera and the script returned:
pi@pointshoot:~ $ python camtest.py
testc
pi@pointshoot:~ $

Not sure if this is what you expected. I would be glad to run some more scripts if you need them.
Thanks,
Danny

tvoverbeek, 2020-07-04:
Thanks Danny. Not the string I expected, but I do not have a HQ camera to check.
But it could be used to distinguish the V1, V2 and HQ camera in the point shoot camera.

PS I'll post my python3 updates to Github in the next few days, so you can get an on-screen battery indicator.

rockman55, 2020-07-20:
tvoverbeek: Have you been able to update the on-screen battery indicator?
Thanks,
Danny

tvoverbeek, 2020-07-20:
Yes, the updates are on github. See commit https://github.com/PiSupply/PiJuice/commit/a2d0d7e2813833108e11b17634ee9bcaf1762baf

rockman55, 2020-07-20:
Can I start with a fresh Buster OS image? Also I should follow the instructions at https://github.com/PiSupply/PiJuice/tree/master/MakerKits/Point-shoot-cam?

Thanks again,
Danny

Sent from Danny Martin

From: Ton van Overbeek
Sent: Monday, July 20, 2020 2:17 PM
To: PiSupply/PiJuice
Cc: rockman55; Mention
Subject: Re: [PiSupply/PiJuice] Makerkit Point-Shoot-Cam is Python 2 only(#514)

Yes, the updates are on github. See commit a2d0d7e

rockman55, 2020-07-21:
When running the install using a fresh image of Buster OS with the Auto-Installation I get the following error:
 remote: Total 3875 (delta 81), reused 97 (delta 37), pack-reused 3710
Receiving objects: 100% (3875/3875), 123.73 MiB | 4.69 MiB/s, done.
Resolving deltas: 100% (2141/2141), done.
Checking out files: 100% (477/477), done.
python3: can't open file 'setup.py': [Errno 2] No such file or directory
start_x=1
start_x=1
bash: line 15: pass: command not found
Remove the HDMI cable and reboot the Raspberry Pi
I have also tried to do the Manual install and still run into problems. Can you give me the sequence of answers I should input after running the step 7.
Install the Media Center HAT software:
cd Media-Center-HAT/Software/
sudo bash install.sh
Thanks,
Danny

tvoverbeek, 2020-07-21:
@rockman55 Sorry, ny mistake. Forgot setup.py in the commit. Now it is there (commit https://github.com/PiSupply/PiJuice/commit/979f24c822c4f9a8d05f5e3071db5d7fab5e1a2f).
Try again.
[image]

rockman55, 2020-07-21:
Still getting errors on Auto Install:

Checking out files: 100% (478/478), done.
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --instal-platlib not recognized
start_x=1
start_x=1
bash: line 15: pass: command not found
Remove the HDMI cable and reboot the Raspberry Pi
pi@pointshoot:~ $ exit

Is there something still missing?
Thanks

tvoverbeek, 2020-07-21:
@rockman55 Thanks for the perseverance in testing this.
There was a spelling error in install.sh (instal-platlib instead of install-platlib).
Also fixed the error on line 15 (pass command not found).
Now I believe it should work with the last commit https://github.com/PiSupply/PiJuice/commit/9476486b8fbdb5ad338ef787ee2a57279807a1b2

rockman55, 2020-07-21:
I just took another step to fix this. After running the Auto Install I had to run steps 2, 3 and 4 from the manual setup process. The camera now works and I can see the battery indicator.
Thanks for your help.
Danny
PS One more additional request for using the HQ camera in the point-shoot-cam.py program line number 339, this is what I changed  [(4056, 3040), (320, 240), (0.0, 0.0, 1.0, 1.0)],  # Large. This sets the resolution to the max for the HQ Camera
Thanks

tvoverbeek, 2020-07-21:
@rockman55 Yes, I know. The code was written (copied from Adafruit) when only the V1 camera existed.
I plan to update it to detect either V1, V2 or HQ and show options accordingly.
Also the use of the crop values does not make sense and I will probably remove them.
Thanks again for pointing out all my mistakes.
Closing this  issue.

closed 2020-07-21 by tvoverbeek

rockman55, 2020-07-21:
Thanks for all your help I will be watching for further updates.
Danny

Sent from Danny Martin

From: Ton van Overbeek
Sent: Tuesday, July 21, 2020 12:57 PM
To: PiSupply/PiJuice
Cc: rockman55; Mention
Subject: Re: [PiSupply/PiJuice] Makerkit Point-Shoot-Cam is Python 2 only(#514)

@rockman55 Yes, I know. The code was written (copied from Adafruit) when only the V1 camera existed.
I plan to update it to detect either V1, V2 or HQ and show options accordingly.
Also the use of the crop values does not make sense and I will probably remove them.
Thanks again for pointing out all my mistakes.
Closing this issue.