PiSupply/PiJuice #515: PiJuice won't reboot my pi rare occasions

kiilez, 2020-07-02:
I am using a PiJuice board in a project. The pi has to be mobile and is in a small case. It takes temperatures and reads RFID boards. The project has been an overall success until recently. I have an external button attached to the j5 header. I have a python gui that creates a process to monitor the button press events using the pijuice API. When a SINGLE_PRESS event occurs it calls a shutdown procedure and calls the 
`pijuice.power.SetPowerOff()`
method with a 10 second delay. I then have the gui shutting the pi down safely. This works flawlessly and then the Pi loses power moments later. Saves the battery from getting used while the pi is halted. This method has worked flawlessly to shut down the pi. 

However, I have had three instances where after this occurs, the pijuice won't start the pi back up. In my configuration for a "PRESS" event I have a "HARD_FUNC_POWER_ON,0" configured in order to apply power back to the pi and boot it back up. This works 99% of the time. On occasion the pi won't boot back up. The PiJuice seems to act like it worked, but the pi just stays off. I have to remove the battery and reattach it in order for it to start back up.

Any ideas on why this could be occurring?

tvoverbeek, 2020-07-03:
Seems you are sorting it out with @ryanteck on Discord. So closing it here.

kiilez, 2020-07-03:
Actually Ryan was not able to fix this. I think it's an issue with firmware
on the pijuice

tvoverbeek, 2020-07-03:
@kiilez A few remarks which might help.

In your first post you have set "HARD_FUNC_POWER_ON,0".
The parameter is 0, the default is 800. The parameter specifies the time in milliseconds in which the button
should be released again to detect a single press.
I do not know how the firmware reacts when this parameter is 0.

A timeout of 10 seconds for the shutdown might be too short. If the power disappears before the shutdown is complete
it might leave the Pi in an undefined state.

There are ways to disable the activity and power leds. Just google for 'RPi disable leds'

kiilez, 2020-07-04:
So, the hard_func_power_on is set by the press event and that 0 parameter can't be changed. My single_press event is monitored by the Python program and is set to a user_event, 800. This way when the button is pressed the Python script can use the API and call pijuice.power.SetPowerOff(10). 

I agree we can probably increase the delay, but this occurs every time I press the button and a shutdown is started then press it again in succession. I think the second button press is putting the pijuice In a strange state because the pi has no power at this point, but if I try to press the button again the pijuice won't power the pi back on. The only way I have found to get it out of this state is to do a HARD_FUNC_POWER_OFF by holding the button down for 20 seconds or disconnecting the battery.

tvoverbeek, 2020-07-04:
@kiilez You wrote: `by the press event and that 0 parameter can't be changed.`
That 0 parameter **can** be changed both in the CLI and GUI.

kiilez, 2020-07-04:
I have attempted to change it in CLI, but it only lets me change the function type and not parameter. The release and press button events don't seem to allow me to change that parameter. All 4 of the other events do.

tvoverbeek, 2020-07-04:
Sorry, you are right for the Press event (which gets triggered as soon as it detects button down).
I was referring to the SIngle Press event (which is what I use normally) which with the default settings is 
coupled to HARD_FUNC_POWER_ON with a parameter of 800.
Any reason you do not want to use the SIngle Press event.

kiilez, 2020-07-04:
I use the single_press event as well, but it is a user_event that my Python
GUI watches for. When the Python GUI sees the single_press event occur it
shuts the pi off. This is so I can have "single press" turn the pi on and
turn the pi off. This is a handheld device and it was deemed that the
operators would be confused if they had to use different types of presses
to power off and on.

tvoverbeek, 2020-07-06:
@kiilez I did some testing over the weekend and believe I have a solution for you.

The firmware polls the buttons in a loop.
When it detects button down it first sets the PRESS event and starts a timer.
If the button is released within the single press time it sets the event to SINGLE PRESS.
When you have defined a function for both PRESS (HARD_POWER_ON) and SINGLE PRESS (USER_EVENT)
you will always get SINGLE PRESS when the firmware is running normally.
After shutdown when the PiJuice goes to low power mode it changes the clock to a very low frequency (the slow weak blink you see then), but the timers used for button press timing keep running normally.
Then the time between 2 polls of the button state is larger than the SINGLE PRESS time and hence the PRESS event
will be used.

Solution is just before you call the shutdown from your USER_EVENT processing change the SW1 button configuration
so the SINGLE_PRESS function is set to NO_FUNCTION.
When you start your user-event program after boot change the SW1 button configuration again to set USER_EVENT as the function for SINGLE_PRESS.
Below the test program I used which uses this strategy:
```
from pijuice import PiJuice
import time
import os

pj=PiJuice(1,0x14)

def EvalButtonEvent():
   global btConfig
   btEvents = pj.status.GetButtonEvents()
   if btEvents['error'] == 'NO_ERROR':
      print(btEvents['data'])
      ev = btEvents['data']['SW1']
      print(ev)
      if ev != 'NO_EVENT':
         if btConfig[ev]['function'] == 'USER_EVENT':
            pj.status.AcceptButtonEvent('SW1')
            # Set single press function to NO_FUNC
            btConfig['SINGLE_PRESS']['function'] = 'NO_FUNC'
            pj.config.SetButtonConfiguration('SW1', btConfig)
            print(pj.config.GetButtonConfiguration('SW1'))
            pj.power.SetPowerOff(15)
            os.system("sudo poweroff")

btConfig = {}
conf = pj.config.GetButtonConfiguration('SW1')
if conf['error'] == 'NO_ERROR':
   btConfig = conf['data']
   print(btConfig)
   btConfig['SINGLE_PRESS']['function'] = 'USER_EVENT'
   pj.config.SetButtonConfiguration('SW1', btConfig)
print(pj.config.GetButtonConfiguration('SW1'))

while True:
   ret = pj.status.GetStatus()
   if ret['error'] == 'NO_ERROR':
      status = ret['data']
      print(status)
      if status['isButton']:
         EvalButtonEvent()
   time.sleep(2)
````
This will then also work while external power is applied to the PiJuice.

kiilez, 2020-07-06:
This looks very promising. I didn't think of changing the configuration for the button events during boot and when shutting down. Currently, my device is on site for some trialing and I can't access the program. I should have it back in about a week. I will apply this programming change and do some trialing myself! I appreciate the help! 

On a side note, do you think it would be possible to just use a SINGLE_PRESS event for shutting down and powering up? When the GUI loads I could change the event configuration to "USER_EVENT" and then when it's powering down, change the event back to "HARD_FUNC_POWER_ON"?

tvoverbeek, 2020-07-06:
Just using SINGLE_PRESS and swapping functions works (Just tested it with my test program).
Note when you have your USER_EVENT program running make sure the pijuice service (system task) is not running (stop and/or disabled) since the service status queries interfere with yours.