PiSupply/PiJuice #516: COMMUNICATION_ERROR appearing on a RPi 4B

Maximuschar, 2020-07-06:
Hello,

I use the PiJuice with my Raspberry Pi 4. When I switch on my Raspberry Pi 4B, everything works.

However, after a moment of operation, just after launching a C code (which uses the lines SPI0, SPI1 and UART0), a "communication_error" appears and so I have no more feedback from my PiJuice device.

Do you have an idea where the error can come from?

I already added "dtoverlay=i2c-bcm2708" in the /boot/config.txt file

Best regards,

tvoverbeek, 2020-07-06:
If the PiJuice firmware version is 1.4 you do not need the `dtoverlay=i2c-bcm2708` in /boot/config.txt.
If your C code is CPU intensive the Pi4B will vary its clock frequency which could cause communication problems
(also on SPI and serial because of changing clock speeds).
Look at the RPi documentation how to keep the clock speed at a fixed value (https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md)

Maximuschar, 2020-07-07:
> If the PiJuice firmware version is 1.4 you do not need the `dtoverlay=i2c-bcm2708` in /boot/config.txt.
> If your C code is CPU intensive the Pi4B will vary its clock frequency which could cause [...]

Hello, thank you for your response.

I currently have the PiJuice firmware in version 1.6. Indeed, I don't need the dtovelay=i2c-bcm2708.

The problem is that I use the SPI intensively but only for a given time (60s). After this, I don't use the SPI anymore and I always get a "COMMUNICATION_ERROR".

Thanks for the document, I'll have a look at it and see if I find something interesting.

Maximuschar, 2020-07-07:
Hello,

I found the issue. I was configuring the I2C SDA to an output in my C code....

Thanks for your help,

Best regards

closed 2020-07-07 by Maximuschar