PiSupply/PiJuice #519: Communication Error preceding large changes in charge level

PitchBright, 2020-07-09:
I'm seeing intermittent "COMMUNICATION ERROR"s in the PiJuice status applet during battery operation. The error occurs on both discharge, as well as during recharge. The error immediately precedes a drastic change in what the status displays for the battery charge levels.

For additional information the Discord server conversation begins here: https://discordapp.com/channels/480159806896865292/481058330190675979/729071589567627325

(New PiJuice Hat on RP4. Firmware 1.4. Fresh OS, no major tasks being performed on the RPi, outside of VNC Viewing so I can watch and monitor the Battery Level via PiJuice Applet. I've tested this with the stock 1820 Battery, as well as the 12,000mAh battery... same results.)

PitchBright, 2020-07-17:
Tested this with a new PiJuice, and a new RPi 4... and the issue is also present there. That rules out any faults in either of my original hardwares I would think.

ryanteck, 2020-07-18:
To resolve @tvoverbeek suggested to set the core clock to a fixed 500Mhz on the Pi 4 with:
```
core_freq_min=500
core_freq=500
``` 
In /boot/config.txt

Because the Pi dynamically changes clock speeds to save power this can cause issues with reliability of reading over I2C therfore causing this issue, by fixing the clock it seems to have resolved it.

This could however mean that the battery won't last as long from where not everything clocks down at idle like it should.

Should now be resolved so closing the issue.

closed 2020-07-18 by ryanteck