PiSupply/PiJuice #522: Solar Panel Requirements

WirthmU, 2020-07-12:
What are the required specifications (Watt, V) for powering PiJuice with a solar panel?

tvoverbeek, 2020-07-12:
Main requirement is the PiJuice input between 4.2 and 10V
Therefore a regulated panel output will be needed in most cases.
See e.g. the solar panels offered by Pi-Supply for the PiJuice.

What values do you need to be readout?
Easiest is make your own based on the API documented on the software page.
Not every all is documented. For the nitty gritty look at the source for the pijuice Python module.
See also https://github.com/PiSupply/PiJuice/blob/master/Software/Test/pijuicetest.py

shawaj, 2020-10-24:
@WirthmU closing this issue now as haven't heard back from you in a while. Feel free to reply back if you do not agree

closed 2020-10-24 by shawaj