PiSupply/PiJuice #523: Run program upon shutdown

carterhickman, 2020-07-14:
Hi Pi Supply! 
I have a question about how to run a python script upon shutdown. I have a programmed timed shutdown that happens every night but I also will need this program to run upon power loss. I was wondering if someone could give me some pointers on how to call a Python script in the case of either of these events. 
Thank you! 
Carter

tvoverbeek, 2020-07-15:
I understand you can already run this python script at a normal shutdown.
With the PiJuice connect the 'No Power' event to this script.

carterhickman, 2020-07-20:
Thanks, tvoverbeek! Do you have a reference to how to connect a python script to the 'No Power' event? I know how to connect the default "events" but not how to connect a script that I create. If there's some documentation of some sorts I would greatly appreciate a link as I'm having a tough time finding it.

tvoverbeek, 2020-07-20:
In the GUI or CLI goto the System Events page
Enable the 'no power' event and in the dropdown select USER_FUNCn.
On the User Scripts page you can define the link between USER_FUNCn and your script (does not have to be Python,
can be a Bash script or an executable) using the full path e.g. /home/pi/myscript.py
The script/program has to be executable and the owner has to belong to the pijuice group.
The default user (usually pi) automatically belongs to the pijuice group.
See also the Software README (https://github.com/PiSupply/PiJuice/blob/master/Software/README.md)

carterhickman, 2020-07-20:
Thank you so much! I'm back in the office on Wednesday and will try this out then. Appreciate it.

shawaj, 2020-12-05:
@carterhickman did you manage to test this out? Did it work ok for you?