PiSupply/PiJuice #524: I2Caddress one conflict, not I have Communication Error

smomid, 2020-07-19:
Hi, 

I have an Audio Injector Soundcard for my Pi

http://www.audioinjector.net/

When I connect the Audio Injector and the Pijuice the Pi won't start up. The guy from Audio Injector suggested there might be  a conflict on I2c address 1. 

So in the PiJuice Configuration settings I changed the I2c address to 16, sort of just picking a number. After I did that immediatlely the Pijuice lost communication, and now every dialog box says "COMMUNICATION_ERROR" in the PiJuice Settings application. 

I tried disconnecting the Pi juice, restarting and going back to default configuration, but that didn't work :(

So my questions are, 1) is there a way to get the Pijuice back to communicating with the Pi? 
2) what is a safe number to change i2C address 1 to so it will be OK? 

Thanks and I hope I can fix this one.

tvoverbeek, 2020-07-19:
Which Audioinjector card are you using?
I decompiled the various audioinjector  devicetree overlays but could not find an i2c address conflict with the PiJuice.
PiJuice uses 0x14. Do you know which i2c address your audioinjector board uses?

Anyway the software uses a fixed address of 0x14. When changed you get continuous COMMUNICATION ERROR.
You will have to modify the 0x14 address in the following files:

/usr/bin/pijuice_sys.py line 273:
```
    try:
        pijuice = PiJuice(1, 0x14)    <==== Change 0x14 to the selected address
        #pijuice = PiJuice(3, 0x14)
    except:
        sys.exit(0
```

/usr/bin/pijuice_gui.py line 34:
```
try:
    pijuice = PiJuice(1, 0x14)   <==== Change 0x14 to the selected address
except:
    pijuice = None
```

/usr/bin/pijuice_cli.py line 19:
```
BUS = 1
ADDRESS = 0x14   <==== Change 0x14 to the selected address
PID_FILE = '/tmp/pijuice_sys.pid'
LOCK_FILE = '/tmp/pijuice_gui.lock'
```

/usr/bin/pijuice_tray.py line 46:
```
        self.pijuice = PiJuice(1, 0x14)   <==== Change 0x14 to the selected address
```

Let us know how it goes.

smomid, 2020-07-19:
Hey tvoverbeek. 

That totally worked. Thanks!

I have another question. So now for the Pijuice I have address 1 and 2 at 16. Should I change address 1 to something else?

After getting the Pijucie to not have a communication error with this change I can get the Audio Injector Soundcard to work and the Pi to start up. 

I am using the Audio Injector Stereo soundcard. 

Someone from Audio Injector sent me this link, where they say there was a conflict with Pijuice on address 14:

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/audioinjector-wm8731-audio-overlay.dts#L22

I posted my issue on their github too. It's over here:

https://github.com/Audio-Injector/stereo-and-zero/issues/60

tvoverbeek, 2020-07-19:
Leave address 2 at 0x68 (for the RTC).
I2C is a serial bua and if the Audio Injector WM8731 is at address 0x1a (as specified in the overlay) there should be
no conflict with PiJuice address 0x14.
So there must be something else going on. 
But if it works for you with address 0x16 for you then continue using it.

shawaj, 2020-10-24:
@smomid is this all working now for you?

smomid, 2020-10-24:
Hi, Yes I got it all working. I'm just not going to change anything or add any i2c devices for now. Thanks for asking @shawaj I think the issue can be closed unless someone has some questions.

closed 2020-10-24 by smomid