PiSupply/PiJuice #525: Two wire external battery

socketwench, 2020-08-02:
I have a piJuice on a Raspi 3, connected directly with no extension header. Originally, I was using the out of the box battery with the board. This was working for the some 10 hours of run time I had used in the project so far. Today, however, I discovered the battery was no longer detected. I had thought it was a connectivity issue, but I'm getting good connection across the pins when tested with a multimeter. 

I decided to try a lipoly battery I had on hand via the screw terminals. Unlike the ones provided by PiSupply, this one only had a two wire interface (+ and -). While this allowed the system to power on, it doesn't appear to charge no matter what setting I use in the battery profile.

Is it possible for the piJuice to charge a lipo without the yellow/NTC line?

tvoverbeek, 2020-08-02:
First select a battery profile which is closest to your LiPo battery. This to avoid a too large charging current.
Better to choose a profile for a bit lower capacity.
Second check the indicated battery temperature in the CLI or GUI.
Pu the temperature sense on 'NO_USED'. The firmware then assumes a battery temp of 25ºC.
Then it should charge. Of course this assumes a single cell LiPo battery (4.2V when full, 3.7V nominal)

socketwench, 2020-08-02:
Thank you! I'll try that!

socketwench, 2020-08-08:
Unfortunately, even after configuring, the battery still reads NORMAL and isn't charging.

tvoverbeek, 2020-08-08:
Then i am at the end of my suggestions.
Do you have technical info about the battery you are using?
Which battery profile do you have selected?

socketwench, 2020-08-08:
I started with the SNN5843_2300 profile, then upped the capacity to 3600mAh, as the battery itself is rated for 3800mAh. I set the temperature sense to NOT_USED as instructed.

socketwench, 2020-08-08:
Ah ha! Under General, I discovered charging was not enabled. I wonder how that happened...

Now it's charging! Thank you again!

closed 2020-08-08 by socketwench