PiSupply/PiJuice #526: PiJuice Firmware Identification on SD clone

MediaMeasure, 2020-08-04:
Hi Guys,

Just wanted to check something concerning me.

I used the pijuice_cli interface to update the firmware from 1.3 to 1.4. 
The update was successful.

I am building a few devices, so what I do is clone the SD and write the image to new cards for the other devices.

I found it strange that on a new device (with new PiJuice), with the cloned SD image the pijuice_cli thinks it has the latest 1.4 firmware?

Now I'm not certain if some PiJuices in my batch were shipped with 1.4 Firmware or if the the Firmware state is saved onto the SD card somewhere? I'm thinking most likely the latter. 

How can I make pijuice_cli test the firmware version from the actual HAT? so that I can update the firmware correctly again via the cli.

Thanks

tvoverbeek, 2020-08-04:
The reported firmware version is read directly from the firmware on the PiJuice.
Recent PiJuices are delivered with the 1.4 firmware.
Otherwise the gui or cli will prompt you to update the firmware.

MediaMeasure, 2020-08-04:
Ah, now I did not expect that. Nicely done. I assumed all the stock I had was 1.3. 

Thanks Ton for the swift reply.

Regards.

closed 2020-08-06 by ryanteck