PiSupply/PiJuice #527: Shutdown rapsberry pi when sunlight is goi

Leumas23, 2020-08-05:
Hellow everyone!!!
I'm using a raspberry pi with a pi juice and a 6w solar panel and I want to shutdown my pi when the sunlight is gone. I was trying with the code on the image bellow but it doesn't work. Anyway to make this?

[image]

tvoverbeek, 2020-08-06:
When do you run  this script?
How do you detect the sun is gone?
How is it not working? More details please.

I would expect an `os.system('sudo shutdown')` before the pijuice.power.SetPowerOff

Leumas23, 2020-08-06:
1. I run the script when the pi is working. I don't know what you mean exactly.
2. I am trying to do it in base of a low voltaje almost 0 when the sunlight is gone (Input no present on this script)
3. I am trying to adapt the script in here: https://learn.pi-supply.com/make/how-to-run-a-user-script-with-the-pijuice-power-events/ what it does is when the main power supply input (charger - in my case the solar panel) is gone the pi shutdown and when the main power supply input is back the pi power on. 
It works only if I completely close the solar panel. I would like to do it when the input voltage of the solar panel is on a low level.

tvoverbeek, 2020-08-07:
The PiJuice has no measurement of the input voltage which you can query.
You wrote the script works when closing the solar panel.
My suspicion is you have a solar panel with a regulated output voltage.
In that case it will produce a constant output voltage (e.g. 5V) until the sun completely disappears.
Just wait till the no power event triggers.

shawaj, 2020-10-24:
@Leumas23 closing this issue now as haven't had a reply in a long time. Feel free to reply if you do not agree

closed 2020-10-24 by shawaj