PiSupply/PiJuice #528: Pijuice Firmware Upgrade-Showing Error

salimness, 2020-08-05:
apt-get install pijuice-base pijuice-gui

Reading package lists... Done
Building dependency tree       
Reading state iapt-get install pijuice-base pijuice-gui
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  cgroupfs-mount
Use 'apt autoremove' to remove it.
The following packages will be upgraded:
  pijuice-base pijuice-gui
2 upgraded, 0 newly installed, 0 to remove and 143 not upgraded.
Need to get 177 kB of archives.
After this operation, 33.8 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-base all 1.5 [100 kB]
Err:1 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-base all 1.5
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:5d22cdd28dc90e14733d1f081bfeaaf7fce3508aad81e92d577c3f64a01d20fb
   - SHA1:a1a381fdaab55801c37920d66f30a5a688ef019b [weak]
   - MD5Sum:6b91a7f5a8a2c7ce34c8e36e89cfac4a [weak]
   - Filesize:100482 [weak]
  Hashes of received file:
   - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
   - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
   - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
   - Filesize:109 [weak]
  Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
Get:2 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-gui all 1.5 [76.5 kB]
Err:2 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-gui all 1.5
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:4a58b696a60945ff19af4d6f814d3690744e24eaf3a406ac801ece255fb24a0a
   - SHA1:0b7e81926fad3367f0f8898c72eb2b9c88fac7d4 [weak]
   - MD5Sum:b1ffe858bb8f388799fc9f6df0639231 [weak]
   - Filesize:76538 [weak]
  Hashes of received file:
   - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
   - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
   - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
   - Filesize:109 [weak]
  Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
Fetched 218 B in 0s (4,451 B/s)
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/p/pijuice-base/pijuice-base_1.5_all.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:5d22cdd28dc90e14733d1f081bfeaaf7fce3508aad81e92d577c3f64a01d20fb
    - SHA1:a1a381fdaab55801c37920d66f30a5a688ef019b [weak]
    - MD5Sum:6b91a7f5a8a2c7ce34c8e36e89cfac4a [weak]
    - Filesize:100482 [weak]
   Hashes of received file:
    - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
    - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
    - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
    - Filesize:109 [weak]
   Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/p/pijuice-gui/pijuice-gui_1.5_all.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:4a58b696a60945ff19af4d6f814d3690744e24eaf3a406ac801ece255fb24a0a
    - SHA1:0b7e81926fad3367f0f8898c72eb2b9c88fac7d4 [weak]
    - MD5Sum:b1ffe858bb8f388799fc9f6df0639231 [weak]
    - Filesize:76538 [weak]
   Hashes of received file:
    - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
    - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
    - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
    - Filesize:109 [weak]
   Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
nformation... Done
The following package was automatically installed and is no longer required:
  cgroupfs-mount
Use 'apt autoremove' to remove it.
The following packages will be upgraded:
  pijuice-base pijuice-gui
2 upgraded, 0 newly installed, 0 to remove and 143 not upgraded.
Need to get 177 kB of archives.
After this operation, 33.8 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-base all 1.5 [100 kB]
Err:1 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-base all 1.5
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:5d22cdd28dc90e14733d1f081bfeaaf7fce3508aad81e92d577c3f64a01d20fb
   - SHA1:a1a381fdaab55801c37920d66f30a5a688ef019b [weak]
   - MD5Sum:6b91a7f5a8a2c7ce34c8e36e89cfac4a [weak]
   - Filesize:100482 [weak]
  Hashes of received file:
   - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
   - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
   - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
   - Filesize:109 [weak]
  Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
Get:2 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-gui all 1.5 [76.5 kB]
Err:2 http://archive.raspberrypi.org/debian stretch/main armhf pijuice-gui all 1.5
  Hash Sum mismatch
  Hashes of expected file:
   - SHA256:4a58b696a60945ff19af4d6f814d3690744e24eaf3a406ac801ece255fb24a0a
   - SHA1:0b7e81926fad3367f0f8898c72eb2b9c88fac7d4 [weak]
   - MD5Sum:b1ffe858bb8f388799fc9f6df0639231 [weak]
   - Filesize:76538 [weak]
  Hashes of received file:
   - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
   - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
   - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
   - Filesize:109 [weak]
  Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
Fetched 218 B in 0s (4,451 B/s)
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/p/pijuice-base/pijuice-base_1.5_all.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:5d22cdd28dc90e14733d1f081bfeaaf7fce3508aad81e92d577c3f64a01d20fb
    - SHA1:a1a381fdaab55801c37920d66f30a5a688ef019b [weak]
    - MD5Sum:6b91a7f5a8a2c7ce34c8e36e89cfac4a [weak]
    - Filesize:100482 [weak]
   Hashes of received file:
    - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
    - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
    - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
    - Filesize:109 [weak]
   Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
E: Failed to fetch http://archive.raspberrypi.org/debian/pool/main/p/pijuice-gui/pijuice-gui_1.5_all.deb  Hash Sum mismatch
   Hashes of expected file:
    - SHA256:4a58b696a60945ff19af4d6f814d3690744e24eaf3a406ac801ece255fb24a0a
    - SHA1:0b7e81926fad3367f0f8898c72eb2b9c88fac7d4 [weak]
    - MD5Sum:b1ffe858bb8f388799fc9f6df0639231 [weak]
    - Filesize:76538 [weak]
   Hashes of received file:
    - SHA256:21f634e088e155af0888fa2a4e4abb11e4de0734e30e2d6f0d3221211541cb84
    - SHA1:928d48e9f8e6709a84252ca931660b73ff1ef236 [weak]
    - MD5Sum:1940f79c863b517b75ade2e45f817442 [weak]
    - Filesize:109 [weak]
   Last modification reported: Thu, 03 Nov 2016 17:19:05 +0000
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

tvoverbeek, 2020-08-06:
Your local package list is out of date.
Current version of the packages is 1.6, not 1.5.
Try to first run `sudo apt update` before upgrading.

shawaj, 2020-10-24:
Closing this issue @salimness . If you do not agree please feel free to reply back

closed 2020-10-24 by shawaj