PiSupply/PiJuice #530: PiJuice with Solar Panel for a Self-Sufficient Bird Box Camera

mrglobetrotter0088, 2020-08-13:
Hello everyone, 

I'm new on raspberry pi and I'd like to learn more about it. 

At the moment, I do my first and own project with a raspberry pi zero and pijuice zero. I'd like to built up a system to use as a remote bird box camera. I've already assembled everything and flashed motioneyeos. The pi is powered by the pijuice with a battery. I want that the battery gets charged by the pijuice solar panel. 

My question is, what the important settings in the pijuice GUI are. I kind of imagine that I have to set the trigger level on 90% and the threshold on 5%. However, what about the watchdog, wakeup alarm and so on? I don't understand the difference and the purpose of those functionalities at the moment. Furthermore, I want to ask you if there is a possibility to program a shutdown at 8 pm and a startup at 6 am. I do not need the camera at this time, since watching birds then is not that interesting.

All in all I would like to create a self-sufficient system that works with solar and remotely. It would be a pleasure if someone can recommend me the settings in pijuice and give me some advices.

Thanks in advance!

ChristopherRush, 2020-08-13:
Take a look at this Project -
https://learn.pi-supply.com/make/pijuice-naturejuice-project/

mrglobetrotter0088, 2020-08-13:
Thank you, but that does not answer all my questions...

ChristopherRush, 2020-08-13:
All the software functions are documented here
https://github.com/PiSupply/PiJuice/tree/master/Software

shawaj, 2020-10-24:
@mrglobetrotter0088 did you need anything else or can we close this issue?

mrglobetrotter0088, 2020-10-24:
Yes, please close it. Thank you.

closed 2020-10-24 by shawaj