PiSupply/PiJuice #531: Turn on Pi when power is applied to the Pi

carterhickman, 2020-08-14:
Hi PiJuice, 

I am working on an exhibit for a science museum in Colorado. I have an automatic startup (from PiJuice) and an automatic shutdown (crontab procedure). Now I need the Pi to turn on when power is applied to the Pi, just in case something happens and it doesn't turn on from the automatic startup. My thought was that I would set the "Wakeup on charge" setting under "System Tasks" in the software settings to 0%. Is this correct? What else do I need to do? Thank you! 

Carter

tvoverbeek, 2020-08-14:
@carterhickman Yes, that is correct.
Some things to know:
- When the PiJuice boots the pi (in your case the automatic startup by the RTC in the morning) the internal wakeup-on-charge setting in the firmware is disabled.
- When shutting down using one of the SYS functions (e.g. SYS_FUNC_HALT_POW_OFF) the internal wakeup-on-charge will be enabled again with the threshold you set in the 'System Tasks"
I believe this is the case for you when you shut down because of 'no power' during the day.

Your cron job shuts down at the end of the day not using the SYS functions and then wakeup-on-charge will not be enabled.
This should be OK, since you want the Pi to boot in the morning, not during the night.

Edge-case: power goes off before the normal shutdown time and comes back after the normal shutdown time.
In this case the Pi will boot, so as a safety you might want to check what the time is directly after boot  and shut down again when it occurs during the off hours.

shawaj, 2020-10-24:
@carterhickman did this work ok for you?

shawaj, 2020-12-05:
Closing as no reply in quite some time. But perhaps relevant to #580

closed 2020-12-05 by shawaj