PiSupply/PiJuice #532: Repeated Wakeup is not working (anymore)

codeafterdark, 2020-08-15:
I had my RPi waking up 4 time per day 8,12,14,16.  I had configured the PiJuice to do this, honestly it worked flawlessly for approximately 1 month.  I have it wake up, run a python script, shutdown, and completely turn off the Pi.  The last part still works fine.  However, it no longer wakes as specific times. I tried to reconfigure through the PiJuice UI and no go.

I checked the everyday box and put 8,12,14,16 in the hours?  Am I using the wrong format?  Can this not be done?

tvoverbeek, 2020-08-15:
@codeafterdark If you specify multiple hours you habe to use ';' as a separator.
So in your case it should be `8;12;14;16`.
Is your RPi connected to the internet? In that case the alarm wakeup is canceled by the timesyncd after boot when you have the id-eeprom address at 0x50.
Change the address to 0x52 and add the line `dtoverlay=i2c-rtc,ds1307` to /boot/config.txt and reboot.
This loads the driver for the ds1307 which is identical to the ds1339 (simulated in the PiJuice firmware) except it does not have an alarm function. Hence no interference with the multiple wakeups, but you still get the time synchronisation of the RTC with the internet.
HTH

shawaj, 2020-10-24:
@codeafterdark is this working now for you?