PiSupply/PiJuice #534: Will the gui interface handle tacks?

klint-k, 2020-08-20:
Hello:
I was wondering if I used "-s" for after a script would the gui muss that up?

Klint

tvoverbeek, 2020-08-20:
??? Do you mean tags?
You are completely free in naming your scripts.
How would the GUI mess that up?

klint-k, 2020-08-20:
Like:

Some_program.py -s
The -s part I thought was called tack.
In python it is called 'arguments'. Will the guilt handle args.

Klint

tvoverbeek, 2020-08-21:
OK, now I understand. Arguments is the usual term.

In the userfunction panel you **cannot** specify arguments (no `some-progam.py -s`).
However depending on the event which triggers the user function it will be called with arguments specifying the event
e.g. event name and in the case of low battery voltage the actual battery voltage.
See the system task source for the nitty gritty: (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py)

closed 2020-08-27 by klint-k