PiSupply/PiJuice #536: Several problems, but mainly never get full charge

guytas, 2020-08-24:
Hi,

Not sure if I have a defective unit but I have several problems with my pijuice.
I just bought it last week and since then, I'm trying several different combination
to make it work.  Let's start by the most important problem first;  

System used: This is running a Pi 4b without any addition on it and a fairly new
Debian installation. I used a 3.1a power supply and also the Raspberry Pi power supply
5.1v 3000ma. This Pi is used to make test before installing on the other Pis

1. The battery never reach full charge while the Pi is running. As of this result,
the battery is on constant charging.  I beleive this is not so good for the battery.
Of course, I'm using the power from the PiJuice because it the only way the power
can be completly cut off from the Pi in the event of a power failure. 

2. Battery percentage level very unstable.  I cannot rely of the percentage it shows
because it can change drastically within a minute or so. For instance, after a
COMMUNICATION ERROR (point 5), the percentage was 72%.  I cannot say how much it
was before the communication error but the 72% sounds too little. anyway, 2 minutes
later it shows 98%

3. One time, after I used up the battery completly, I connected back the power on the
piJuice and went away. Two days later, the battery was only at 81%.

4. For testing, I had set the minimum change to 40% and the "wakeup on charge" at 55%.
Started the test at 95%. Keep in mind that this whole test last 2 minutes.  OK, I started
to unplug the power supply.  Everything was OK. Reconnect the power supply.  Still OK.
Removed the battery. The system started to reboot constantly. Put back the battery.
The battery percentage was then 45% (4.16 volts).  I dont beleive that the percentage
of the battery was really 45%. But this is still the value the system will use to controls
the "system events".

5. COMMUNICATION ERROR. This one is kind of hard to understand with a so short line
between the hat and the Pi. On the first day I fire it up, I was getting COMMUNICATION
ERROR every 3 or 4 minutes.  I was trying to set some options in the HAT but the
error message kept poping up.  Now, 3 or 4 days later, I get COMMUNICATION ERROR
every now and then.  May be 2 or 3 times per evening. STill...

6. Unstable battery temperature.  Looking at the temp, after I've been using the Pi
for a couple of hours, the temperature was showing 24.  Impossible as the battery felt
quite warm.  But then, as I was watching it, the temperature changed to 46,  then later
24, then 46 again

7. SW2 is not working.  I can program SW1 and SW3 but SW2 never react to any command.

8. Programming the switches is very difficult.  I still dont understand when the function
are really recorded.  I explain;  I change SW3 option for SINGLE PRESS to use
SYS_FUNC_HALT_POW_OFF for example.  I try to press the button, nothing is happening.
I closed the window and re-open; nothing.  I rebooted and try again; nothing. But then
later on, when I try the SW3 for single press, it works.

The weirdest is that I reprogrammed this same button for HARD_FUNC_POWER_ON.  Saved and
closed the window. Tried the SW3 and the old function is the one that gets executed.
i.e. it executed the POW_OFF.  So this is why I say that I dont understand WHEN the
real function is saved.

9. I'm using the system task "Minimum charge" to turn off the Pi before the battery dies.
But I understood that it was possible to replace the battery without any down time. But
I guess this is conflicting with the minimum charge. Whan I tried it, the Pi kept
rebooting all the time.  So I checked the option "no battery turn on" thinking the 
the event of "no battery" will take precedence over the "minimum charge" but it didn't.

10. Starting up the Pi without the PiJuice is causing the PiJuice icon to be blank. You can
select the settings by cliking on the blank icon fine but the icon is not showing.

tvoverbeek, 2020-08-24:
Before trying to comment on your observations I would like to know the OS (Raspbian/Raspberry Pi OS version, or ...),
the installed version of the pijuice-base and pijuice-gui packages and the version of the PiJuice Firmware (visible on the
Firmware tab in the GUI/CLI).
Your fluctuating charge levels and communication errors smell like an older version fo the firmware.
Actual firmware version is 1.4

guytas, 2020-08-24:
Omg i forgot to mention that. Sorry about that.

The pijuice is 1.4. It did update itself when i first installed it.  I also do not have the dtoverlay=i2c-bcm2708 in the config.txt but I dont think it matters anyway.

The os is raspbian version 10

Thanks

tvoverbeek, 2020-08-27:
@guytas Sorry for not answering earlier.

Firmware and OS are OK. With the 1.4 firmware you should not use the `dtoverlay=i2c-bcm2708` in /boot/config.txt.
The communication errors you are seeing could be caused by the Pi4 changing its core frequency. It goes down when not very busy and up when it has lots to do. The I2C clock is derived from the core frequency.
To force a fixed core frequency (and a constant I2C clock frequency) add the following lines to /boot/config.txt:
```
core_freq=500
core_freq_min=500
```
and see if the communication errors disappear. The changing I2C clock frequency could also have caused your varying
charge level and temperature readings.

See if this helps, before we attack the other issues.

guytas, 2020-09-02:
Ok, yes with these parameters, the comm error seems to be gone. But really my point 1 is the one that bugs me the most.  And if this can give you a better clue... read this...

I connected a SSD drive on USB 3. The power for the ssd also comes from the USB. The power on idle is something like 0.170amp. I don't have the exact value with me but I remember checking it and it was around that.  But after about half a day, the power of the battery dropped from 97 to 92%.  It's decreasing instead of getting to its full 100%.

Hope this gives you a better clue.

Thanks

Guy

tvoverbeek, 2020-09-02:
Your combination of RPi4 + peripherals is at the limit or over the limit of what the PiJuice can supply via the GPIO 5V.
I would connect your primary 5V supply to the Pi and not to the PiJuice.
You can still remove power to the Pi in the case of a 5V power interruption.
Use the 'no power' event to shut down the pi (Pijuice now supplying the 5V) and remove the 5V from the PiJuice to the Pi.
When 5V returns the Pi will boot again.
Any excess power from the power supply will then be used to charge the battery.
Battery charge level is an estimate. The charge control IC works independently from the charge level estimation.
You could suspend charging when the level is say above 95% and resume when the level drops below say 50%.

guytas, 2020-09-13:
Ok, yes that seems to be better in the way that it stops charging. But how do I do this:

« You could suspend charging when the level is say above 95% and resume when the level drops below say 50%.

I couldn’t find how

tvoverbeek, 2020-09-14:
@guytas You need a user program periodically checking the charge level (pijuice.power.GetChargeLevel()).
When over say 95% disable charging (pijuice.config.SetChargingConfig({'charging_enabled': False}))
Of course you need to enable charging again when the chargelevel is below a threshold.
You could use the 'low charge' event and trigger a user function which enables charging again:
pijuice.config.SetChargingConfig({'charging_enabled': True}).