PiSupply/PiJuice #538: Battery not enough for Raspberry Pi 4?

Masinoha, 2020-08-28:
Hello. I have an issue with the Pijuice I have with regards to using it as a UPS for the Raspberry Pi 4.

My peripherals are as follows: one USB wireless mouse, one USB headset, one fan HAT ([link](https://www.lazada.com.ph/products/raspberry-pi-led-gpio-extension-board-dual-cooling-fans-for-raspberry-pi-4b3b-accessories-i891282282.html?fbclid=IwAR0S22G5rJcPAEymEwHILDOzmKlfFqGWKJmkc32ckjI0Tr0vhkLx_jj_t5g)), and finally the Pijuice attached on top. I am also using the official the official 7" screen and the official PSU for Raspberry Pi 4 which supplies 5V and 3A. I think it would also be important to note that the **Raspberry Pi 4 and the screen are both connected to the power supply using a USB-C splitter cable** (I am using the [SmartiPi Touch 2 ](https://smarticase.com/products/smartipi-touch-2)case ). The battery I'm using is the BP7X that came with the Pijuice.

I have the Pijuice Settings software installed and have updated the Pijuice firmware to 1.4

I also have over_voltage=6 at my config.txt. I am using the Raspbian Buster.

**HOW IT USED TO WORK**

My typical set-up for using the Raspberry Pi involves using the peripherals mentioned above simultaneously, with the GPIO fan HAT's fan and LED turned on. I tested the Pijuice before if it could keep the Pi on (No programs running) by unplugging the PSU. It worked before and the Pi, the fan HAT, and the screen were still running despite being powered from the Pijuice. Though the battery drained relatively quickly (about 1% every 20 seconds), it still provided enough time for initiating a safe shutdown. I thought that I had set up the Pijuice well enough to work as a UPS during power outages.

To summarize, it did work as a UPS when I first tried it (and several attempts after that)

**THE PROBLEM**

After quite some time, we had a power outage in our area. Instead of the Pi acting as a UPS, it merely stayed on for five seconds then just shutdown completely. It's worth noting that when this happened, the **Pijuice itself also seemed to turn 'off' as the green flash from LED1 that momentarily blinks from the Pijuice no longer flashes until I plug the PSU back in**. I tested this again. Booted the Pi by plugging in the PSU and waited for the desktop to load. I checked using i2cdetect  -y 1 and confirmed that addresses 14 and 68 are occupied. Checked the Pijuice software and everything seemed to be okay or similar to as they were during testing. I closed all running applications and unplugged the PSU. The result was that the Pi immediately shutdown along with the Pijuice itself (no green blinking from LED1 even a minute after the plug was pulled)

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    

The only recent change I remember doing to the software was just halving the long press lengths for SW1. If I remember correctly, everything else in the software side is on default.

The final tests I've been doing is to simply start the Pi through SW1 while unplugged from the PSU. The Pi boots up and screen and fan start, however, as the screen approaches to the point of displaying the desktop, the Pijuice and the Pi again shutdown. 

Reading up on this, I have started to suspect that the problem lies in current draw? Maybe the standard BP7X cannot supply enough for the peripherals to work? But then it would contradict my first time tests as it was able to act as a UPS for the Pi even with all those peripherals attached. Could it be damage to the Pijuice or the battery then? Honestly, I've run out of ideas and would like to have some assistance and guidance for troubleshooting this.

**Additional Notes**

-According to the software, the battery charge never goes beyond 92%. Once it reaches that number, it would just say NORMAL instead of charging from GPIO
-When I unplug the PSU now, the Pi shutdowns but the Pijuice seems to shutdown as well (no green blinks from LED1), pressing SW1 does nothing and the Pijuice seems to only turn on when I plug the PSU back on
-I suspect it to be a current issue due to the number of peripherals I have attached but why then did it work decently the first time?
-According to Pijuice Settings, GPIO Power Input seems to present a negative current most of the time, what could that possibly mean?
-In the Pijuice HAT Configuration window, under the Battery tab, the profile listed is PJLIPO_12000 instead of BP7X_1820. Will try to change this later and update the post

I still haven't tested the Pijuice without any other peripherals attached to the Pi. I will update this post with the results once I get around to doing it.

I do apologize if my post does not contain much relevant information or is incomplete as I felt that if I continued testing aimlessly, I would just be wasting time or possibly damage the HAT or Pi. This is also my first post ever in Github and my first time ever asking online for support. If I lack the necessary information in this post, tell me and I will do what I can to provide it xD Thank you for your time and your guidance and input is greatly appreciated!

tvoverbeek, 2020-08-28:
Your Pi with all the peripherals and overvolting will most likely use more current than the BP7X can deliver, especially if it is not full.
Max current the PiJuice can deliver is 2.5A
Also the 12000mAh profile will cause problems. For the 12000 the constant charge current is 2.5A while for the BP7X it is 0.925A.
Termination current (when charging stops) is 350mA for the 12000 while for the BP7X it is 50 mA.
You may have accidentally changed the dip switch setting on the PiJuice which determines the default battery profile when not specified by the user in the GUI/CLI.
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#dip-switches-hat-only

Recommendation: Fix the battery profile so it matches the battery and use a larger capacity battery for your setup.

Masinoha, 2020-08-28:
> Your Pi with all the peripherals and overvolting will most likely use more current than the BP7X can deliver, especially if it is not full.
> Max current the PiJuice can deliver is 2.5A
> Also the [...]

It seems that I did flip the switch by accident before. The profile matches now.

Tried to boot without the screen. The Pi boots just fine and all. It really is a current issue, I guess. A larger capacity battery really is the solution to this. It's just that getting peripherals and HATs for the Raspberry Pi here in Southeast Asia is pretty difficult. I guess there's no choice but to get a 12000mAh battery from PiSupply itself. Well anyway, thank you for the swift reply :)

closed 2020-08-29 by Masinoha