PiSupply/PiJuice #540: Communication_error while using pijuice zero

JoshYang1977, 2020-09-02:
Hello,

I have a communication_error while using pijuice zero.
I used  i2cdetect -y 1 and found nothing.
Already used all other methods mentioned in other posts.

Any suggestion? I use Pi Zero to install the pijuice zero
thank you very much.

Josh

JoshYang1977, 2020-09-02:
I just found an error occurs

i2c-dev not found in dictionary /lib/modules/4.19.97+

any suggestion? thanks.

JoshYang1977, 2020-09-02:
now I refresh os and i2c-dev not found in dictionary /lib/modules/4.19.97+ does not occur.
However, i2cdetect -y 1  still finds nothing. Below are pictures to show the current status.

thanks

[image]
[image]
[image]

ChristopherRush, 2020-09-02:
How have you physically connected the board? Do you have a photo?

JoshYang1977, 2020-09-02:
Chris,

Thank you very much for your response. I uploaded the pictures in my previous comment.
Finally, I figured that out and make it works. Basically, the pi zero is something broken and I have it installed in a new one. Now it is working. Thank you very much.

Josh

closed 2020-09-02 by JoshYang1977