PiSupply/PiJuice #542: Shutting Down

Seth-AIG, 2020-09-09:
I'll explain my needs and maybe you can suggest. I'm using this RPI in a drone setup. The power switch activates the RPI and a few other things. When I shut off the power, I want to use the PiJuice to simply continue to supply power while having the RPI gracefully shutdown. I tried SYS_FUNC_HALT_POW_OFF and it does shut down after 60 seconds like it should. I read some support pages that stated I could opt for a user script to get that down to 20 seconds. I'm new at this and wondered if someone could supply me with that entire script?

Second issue I have found is that when the RPI is plugged into the USB C wall adapter, everything powers, functions and shuts down after 60 seconds like it should. When I switch over to using a DC to DC buck converter and a 6S lipo battery, it boots, shows the desktop and then PIJuice kicks in and shuts everything down but not completely. It eventually gets to a black screen and shows the same message but with two different numbers before it. The only way to get the RPI to boot again si to remove the PIJ battery.

66.911690 reboot system halted
86.332803 reboot system halted

tvoverbeek, 2020-09-10:
For your first question:
Here is a user script in python:
```
#!/usr/bin/python3

import os
from pijuice import PiJuice

pj = PiJuice(1, 0x14)
# Set power off time-out to 15 seconds (or shorter if your Pi is shutting down faster)
pj.power.SetPowerOff(15)
# Now shutdown
os.system("sudo poweroff")
```
Name the script e.g. myshutdown.py and make it executable
Now in the GUI/CLI System Events page connect the 'no power' event to e.g. USER_FUNC1
On the User Scripts page define the full path to your myshutdown.py (e.g. /home/pi/myshutdown.py) as USER_FUNC1

For your 2nd question.
How is the output from the buck converter connected to the Pi/PiJuice?
It smells like the Pi is shutting down twice 20 seconds apart due to undervoltage(?).
Is the output from the converter a constant 5V or is it varying depending on the power drawn by the rotors?

Seth-AIG, 2020-09-10:
How is the output from the buck converter connected to the Pi/PiJuice?
Everything runs through the USB C of the RPI4. Then the PIJuice is powered via GPIO.

It smells like the Pi is shutting down twice 20 seconds apart due to undervoltage(?).

Is the output from the converter a constant 5V or is it varying depending on the power drawn by the rotors?
Yes, its getting a constant 5.09v. Ive got the buck set to 5.12 and with all the draw it bumps down to 5.09/5.08.

Seth-AIG, 2020-09-10:
So I ran through the script settings, did sudo chmod +x myshutdown.py and when I power off, it doesnt do what is expected.

tvoverbeek, 2020-09-10:
User scripts cannot be run as root. You can do the chmod as normal user (pi, I suppose).
Moreover the user has to belong to the pijuice group. The package installation will add the default user (uid 1000, normally pi) to the pijuice group.
Does the script work when you execute it directly if the Pi is powered by the PiJuice (after of course disabling the  'no power' event)?

aerdia, 2020-09-10:
When I tried to run the chmod without sudo, it said I didn't have permission. I'm still learning, so you may need to spell it out a bit for me please. 

How do I try to run it from the command line? ./myshutdown.py?

tvoverbeek, 2020-09-10:
Did you create the script with sudo?
Then the script is owned by root.
Are you running Raspbian (now Raspberry Pi OS) or an other OS?
Are you logged in as normal user (like pi) or as root?

aerdia, 2020-09-10:
Running on a pi4 with Ubuntu 18.04. It's under the user I created "Ubuntu" that has admin rights. How do I switch to the pi user?

tvoverbeek, 2020-09-10:
That makes a difference.
Are you running the 32-bit or the 64-bit version?
Have you followed the installation procedure in https://github.com/PiSupply/PiJuice/tree/master/Software/OS%20Support for your Ubuntu version? Note the steps to add the user to the i2c and pijuice groups.

Seth-AIG, 2020-09-10:
64 bit and I followed your instructions.

Seth-AIG, 2020-09-10:
This is what I followed. I dont see anything about users?

https://github.com/PiSupply/PiJuice/blob/master/Software/OS%20Support/ubuntu_server_64.MD

ChristopherRush, 2020-09-10:
Ton,

I have been dealing with Seth on a support ticket to help him get up and
running on Ubuntu Mate 64-bit. So far he has the software running and has
added the user to both i2c and pijuice groups. I have closed the support
ticket to avoid duplicating our efforts has a GitHub issue was opened
after.

tvoverbeek, 2020-09-10:
@Seth-AIG OK. Let us take it one step at a time.
Your are logging in as user Ubuntu.
Does it have administrative rights without using sudo?
Can you run `i2cdetect -y 1` without sudo?

Seth-AIG, 2020-09-10:
Yes.

ubuntu@LiDAR_Kit:~$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
ubuntu@LiDAR_Kit:~$

Seth-AIG, 2020-09-10:
Not sure why the strike through but that wasnt on my screen.

shawaj, 2020-09-10:
@Seth-AIG just a formatting thing in markdown. Fixed it now

tvoverbeek, 2020-09-11:
@Seth-AIG This looks fine.
0x14 is the I2C address for the PiJuice, 0x68 is the simulated RTC in the PiJuice firmware.
So user ubuntu has normal I2c access.
Can you show the output from `cat /etc/group | grep ubuntu`.
This to check if user ubuntu is a member of the groups i2c and pijuice.
Also check if the pijuice service is running: `systemctl status pijuice.service`

Seth-AIG, 2020-09-11:
ubuntu@LiDAR_Kit:~$ cat /etc/group | grep ubuntu
adm:x:4:syslog,ubuntu
dialout:x:20:ubuntu
cdrom:x:24:ubuntu
floppy:x:25:ubuntu
sudo:x:27:ubuntu,development
audio:x:29:pulse,ubuntu
dip:x:30:ubuntu
video:x:44:ubuntu
plugdev:x:46:ubuntu
lxd:x:108:ubuntu
netdev:x:111:ubuntu
ubuntu:x:1000:
i2c:x:133:pijuice,ubuntu
pijuice:x:1001:ubuntu

ubuntu@LiDAR_Kit:~$ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: 
   Active: active (running) since Thu 2020-09-10 09:20:54 EDT; 21h ago
 Main PID: 836 (pijuice_sys.py)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/pijuice.service
           └─836 /usr/bin/python3 /usr/bin/pijuice_sys.py

Sep 10 09:20:54 LiDAR_Kit systemd[1]: Started PiJuice status service.

Seth-AIG, 2020-09-11:
I also tried to run the .py script.....

ubuntu@LiDAR_Kit:~$ python myshutdown.py
Traceback (most recent call last):
  File "myshutdown.py", line 4, in <module>
    from pijuice import PiJuice
ImportError: No module named pijuice

ChristopherRush, 2020-09-11:
The script must be run as Python3

tvoverbeek, 2020-09-11:
The groups info and the systemctl output all look as they should be.
I suppose the default python on Ubuntu is still python 2.7.
The pijuice module is only available for python 3.x. (Python 2 is not maintained any more).
So try your script with python3 myshutdown.py

Seth-AIG, 2020-09-11:
So the screen ran and immediately when to the black shut down screen but after 15 seconds it did not shut off the RPI power (Red and Green Lights).

tvoverbeek, 2020-09-11:
The power removal only works when the PiJuice is the only 5V source for the Pi (via the GPIO).
If you somehow have 5V on the Pi micro-USB it should shutdown, but the power is not removed.
Does the green led stay on? Normally at the end of the shutdown it flashed 10 times and then goes off (provided it has the normal activity function).
If not we have to investigate why the shutdown is not working.

Seth-AIG, 2020-09-11:
So I shut off the USB C power, PI on PiJuice power and then ran the script and it ran exactly as it should. Here are my pij settings.

[image]
[image]
[image]
[image]

Seth-AIG, 2020-09-11:
System Task>>System task enabled
System Events>>No power:>>USER_FUNC1
User Scripts>>USER FUNC1:>>/home/ubuntu/myshutdown.py

myshutdown.py
#!/usr/bin/python3

import os
from pijuice import PiJuice

pj = PiJuice(1, 0x14)
# Set power off time-out to 15 seconds (or shorter if your Pi is shutting down $
pj.power.SetPowerOff(15)
# Now shutdown
os.system("sudo poweroff")

tvoverbeek, 2020-09-11:
So the script works.
Now why does the script not work on the 'no power' event.
The script should have run automatically when you removed the USB-C power.
What does `ls -l /home/ubuntu/myshutdown.py` show?

Seth-AIG, 2020-09-11:
ubuntu@LiDAR_Kit:~$ ls -l /home/ubuntu/myshutdown.py
-rwxr-xr-x 1 root root 235 Sep 10 09:34 /home/ubuntu/myshutdown.py

Seth-AIG, 2020-09-11:
Looks like I need to change the ownership to ubuntu?

chown ubuntu myshutdown.py?

tvoverbeek, 2020-09-11:
Aha, that is the problem. script is owned by root. The pijuice service only executes user scripts where the owner is not root (security reasons, script could take over the machine) and the owner has to belong to the pijuice group.
Yes, change the owner to ubuntu

Seth-AIG, 2020-09-11:
ubuntu@LiDAR_Kit:~$ sudo chown ubuntu myshutdown.py
ubuntu@LiDAR_Kit:~$ ls -l /home/ubuntu/myshutdown.py
-rwxr-xr-x 1 ubuntu root 235 Sep 10 09:34 /home/ubuntu/myshutdown.py

tvoverbeek, 2020-09-11:
Just to be sure also change the group to ubuntu:
`sudo chown ubuntu:ubuntu /home/ubuntu/myshutdown.py`

Seth-AIG, 2020-09-11:
It works! Ok, if we can get the DC power issue worked out, I will be all set.

Seth-AIG, 2020-09-11:
I just tested on DC power and its still doing the same. I was hoping the other fix was apart of this problem as well somehow.

tvoverbeek, 2020-09-11:
I am afraid I cannot be of much help with your DC-DC converter.
As far as I know a buck converter is a switching power supply and will have an output voltage ripple.
Maybe the ripple is too large for the Pi.

Seth-AIG, 2020-09-11:
With the Pijuice not on the hat, it works perfectly.

Seth-AIG, 2020-09-14:
Checking back to see if you may have some additional suggestions? When I remove the PiJuice and use the RPI with the DC-DC converter, it works as it should. Add the PiJuice and it forces the unit to reboot and it halts.

tvoverbeek, 2020-09-14:
It could be that the PiJuice firmware is in a funny state. You can reset it by pressing SW1 and SW2 (note SW2 position is at the side, not in the middle) simultaneously for 20 seconds. During the reset both LEDs will blink alternating red and blue.
I just had one of my PiJuices causing a shutdown 20 seconds after boot. Resetting the PiJuice solved it.

Seth-AIG, 2020-09-15:
Can you point me to something that labels which is SW1 and 2?

shawaj, 2020-09-15:
@Seth-AIG https://github.com/PiSupply/PiJuice/tree/master/Hardware

Seth-AIG, 2020-09-15:
After doing as suggested, when it booted the LED was red and in the GUI it showed the battery charging state as bad. Seems there is an issue with this unit. I purchased through Amazon but it appears there are no more to do an exchange. Do you have more stock elsewhere?

Seth-AIG, 2020-09-17:
Any recommendations before I get a refund from Amazon? I really wanted this to work.

tvoverbeek, 2020-09-17:
@Seth-AIG Sorry, I am out of suggestions and recommendations.

shawaj, 2020-10-24:
@Seth-AIG did you get this resolved in the end?

Seth-AIG, 2020-10-24:
No. I had to return the unit sadly. 
    	
    	Get Outlook for iOS

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/PiSupply/PiJuice/issues/542#issuecomment-715645749",
"url": "https://github.com/PiSupply/PiJuice/issues/542#issuecomment-715645749",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

shawaj, 2020-10-24:
@Seth-AIG I don't suppose you have a link to the DC-DC converter you were using? And the 6S battery?

Also, how did you have them wired up if you can remember?

Seth-AIG, 2020-10-25:
> @Seth-AIG [1] I don't suppose you have a link to the DC-DC converter you were using? And the 6S battery?

Here is the converter. The battery wasn't the issue as it happened with
multiple batteries as well as I used an AC to DC converter also. I would
love to get this issue fixed. If you need someone to beta test it for
you and you want to send me a testing unit, I would be open to that. 

https://www.amazon.com/gp/product/B079N9BFZC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

 

Links:
------
[1] https://github.com/Seth-AIG
[2]
https://github.com/PiSupply/PiJuice/issues/542#issuecomment-715671159
[3]
https://github.com/

shawaj, 2020-10-25:
@Seth-AIG how did you connect the converter? With a micro USB connector or some other way?

Seth-AIG, 2020-10-25:
I used a jst end on the converter to a usb c connector that connected to the raspberry pi 4. 
    	
    	Get Outlook for iOS

shawaj, 2020-11-03:
What happens if you use a microUSB connected directly to the PiJuice HAT input instead?

Seth-AIG, 2020-11-03:
I didn’t try it and no longer have the unit as it was sent back for refund. 
    	
    	Get Outlook for iOS

tvoverbeek, 2020-11-03:
Since you do not have the unit any more, closing the issue

closed 2020-11-03 by tvoverbeek

Seth-AIG, 2020-11-03:
I’m more than willing to trouble shoot it if you want to send a test unit. Id like to use this in our systems. We are building drone hardware that we would like to use in each one sold. 
    	
    	Get Outlook for iOS

shawaj, 2020-11-03:
@Seth-AIG appreciate the offer but we can't really send one free of charge. If you do decide to purchase one again and need any more help then please let us know

aerdia, 2020-11-04:
I purchased another one. It’ll be here by Tuesday.

Thanks so much,
Seth Zora
President and Lead sUAS Pilot
AerdiA
Work: 724.693.8292
Cell: 412.708.7088
Web: https://www.aerdia.com
Email: zora.s@aerdia.com

IMPORTANT: This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you have received this email in error please notify the system manager. Please note that any views or opinions presented in this email are solely those of the author and do not necessarily represent those of the company. Finally, the recipient should check this email and any attachments for the presence of viruses. The company accepts no liability for any damage caused by any virus transmitted by this email.

shawaj, 2020-11-04:
ok let us know what happens and we can reopen if necessary